edufolly/flutter_bluetooth_serial #2: Feature Request: iOS support

dude8604, 2018-09-23:


edufolly, 2018-11-22:
Sorry, but I'm not an iOS developer. I need someone to help with this.

SouengKimmeng, 2019-03-25:
+ 1 it would be great if it gonna work with iOS device.

joaowd, 2019-05-03:
Yeah I'm also looking for this

gillestasse, 2019-06-18:
I am also very interested. I am not an iOS developer but I know that the Bluetooth device need to be a part of MFi Certification for Bluetooth Classic Device. https://developer.apple.com/programs/mfi/

Yahhi, 2019-07-01:
HC-05 is not compatible with iOS. https://stackoverflow.com/questions/50443456/connecting-hc-05-with-iphone-se-iosv11-0

davidhbrown, 2019-07-25:
I'm not an iOS developer either and Android-only is good enough for my immediate needs. But I did buy a couple of Bluetooth 4 "HM-10 compatible" modules, I have an aging MacBook (I think... haven't turned it on in a couple of months), and I could borrow an iPad. So, if these resources would allow me to help test some future version of the example project on iOS, I'd be happy to lend what help I can. Have the project watched and starred... really appreciate the work you've put into it!

nickroeker, 2020-05-23:
As I have a rather immediate need for this, I'd like to take a stab at it but I'm new to Flutter, Flutter plugins, and iOS development (but thankfully not development in general!). Does anyone have leftover code from taking a stab at this by chance?

If I take this on I also need someone to brutally review it due to my inexperience on all necessary subjects.

CodingAleCR, 2020-07-02:
@nickroeker I can help you out, let me know what do you need, maybe we could tag team this?

coderinblack08, 2020-11-15:
What is the equiv of bluetooth socket in ios?

Moussabaddour, 2021-04-21:
I am trying to connect iPhone with Raspberry Pi, I got this error (flutter: MissingPluginException(No implementation found for method connect on channel flutter_bluetooth_serial/methods))

mbabuglia, 2021-08-03:
What do you mean for iOS support for this plugin?  This plugin is for Classic Bluetooth (SPP profile), iOS doesn't work with SPP, it can only use Bluetooth Low Energy (GATT profile).  So to communicate in iOS you must use other plugins like flutter_reactive_ble or flutter_ble_lib

gillestasse, 2021-10-05:
@mbabuglia  To my understanding, it is possible. It is called Bluetooth Classic. The device has to be registered with the Mfi program through. https://mfi.apple.com/
There are some plugins designed for Cordova : https://github.com/soltius/BluetoothClassicSerial

mbabuglia, 2021-10-05:
@gillestasse , thank you for your links. I researched the availability of Bluetooth SPP in iOS almost two years ago but didn't find anything about MFI.  We sell a hardware device for agriculture industry and have been replacing the Bluetooth module on old devices with a new module with GATT. New devices are all GATT. 

Does anyone have experience with MFI program? Looks that you have to pay 100 usd just to be able to ask preliminary questions to someone on Apple.

Anyway, back to this plugin. Altough MFI is an option, how many on this thread are developers of a hardware device and can thus go through MFI certification? Or are developers needing to connect to an already approved MFI device? 

Seems to me that the need/use for this library in iOS is very limited.

gillestasse, 2021-10-05:
@mbabuglia We are also creating apps for agriculture. We are using bluetooth to enable users to read rfid tags. Sheep farmers are willing to use only android for their daily use. But horse breeders want to stick to their iOs. So it is quite a problem for us. We have not find a BLE device that we can use yet.

Krahos, 2021-11-19:
@mbabuglia I'm working with a sensor developed in house and for the future we would be interested to port our software on iOS (but also on other platforms) and the only thing constraining us is the lack of support for a library that supports multiple platforms.

Satyam953, 2022-01-16:
Hi edufolly, please provide ios support.

mbabuglia, 2022-02-24:
@Krahos, you have other libraries that support mulitple platforms. Your sensor has to work with BLE though.
Does anyone here have experience with MFI program? For the moment we are not going to enroll, seems to be too bureaucratic.

Krahos, 2022-02-28:
> @Krahos, you have other libraries that support mulitple platforms. Your sensor has to work with BLE though.

The problem is that for now our sensor doesn't support BLE, but yeah it seems easier to just work towards that direction if we want to go multiplatform.