edufolly/flutter_bluetooth_serial #6: how to use this plugin (more errors when i use it)

mouradouddah, 2018-12-09:


edufolly, 2018-12-16:
Hi @mouradouddah,
please, provide a new issue with more details about your problems.
You can see the [example](https://github.com/edufolly/flutter_bluetooth_serial/blob/master/example/lib/main.dart) to start to develop your app.

mouradouddah, 2018-12-19:
hi, I eagerly wait for the new version.

edufolly, 2018-12-20:
I published a new version of the plugin. Please, make some tests and if you find some problem open a new issue.

closed 2018-12-20 by edufolly

mouradouddah, 2018-12-21:
Yes I tested your new version (**flutter_bluetooth_serial 0.0..4**) and now I communicated **HC05** with my mobile without errors. thank you very much **edufoly** .
                                                      **we are still waiting for you new plugins and updates**