edufolly/flutter_bluetooth_serial #7: Error while trying to connect to a disconnected printer

rafaelterada, 2018-12-13:
Good Morning. I am using this plugin to connect to a printer and print. It works fine but when I try to print to a paired printer that is turned off, it generates an error. How can I check if the printer is turned off or not? Thanks

   `List<BluetoothDevice> devices = await bluetooth.getBondedDevices();
    if (devices.length > 0) {
      macPadrao = p.getString('impressora');
      if (macPadrao != null) {
        device = devices.singleWhere((d) => d.address == macPadrao);
        if (device != null) {
          try {
            bool conectado = await bluetooth.isConnected;
            if (!conectado)
              await bluetooth.connect(device).then((d) {
                device.connected = d;
              }).catchError((error) {
                print(error.toString());
              });
          } catch (exception) {
            print(exception.toString());
          }
        }
      }
    }`

E/FlutterBluePlugin(10153): java.io.IOException: read failed, socket might closed or timeout, read ret: -1
E/FlutterBluePlugin(10153): 	at android.bluetooth.BluetoothSocket.readAll(Unknown Source:69)
E/FlutterBluePlugin(10153): 	at android.bluetooth.BluetoothSocket.readInt(Unknown Source:3)
E/FlutterBluePlugin(10153): 	at android.bluetooth.BluetoothSocket.connect(Unknown Source:123)
E/FlutterBluePlugin(10153): 	at io.github.edufolly.flutterbluetoothserial.FlutterBluetoothSerialPlugin.lambda$connect$0(FlutterBluetoothSerialPlugin.java:221)
E/FlutterBluePlugin(10153): 	at io.github.edufolly.flutterbluetoothserial.-$$Lambda$FlutterBluetoothSerialPlugin$52C5RUgoP_r70sgAnP8ZXC1Ysz0.run(Unknown Source:6)
E/FlutterBluePlugin(10153): 	at android.os.AsyncTask$SerialExecutor$1.run(Unknown Source:2)
E/FlutterBluePlugin(10153): 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
E/FlutterBluePlugin(10153): 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
E/FlutterBluePlugin(10153): 	at java.lang.Thread.run(Thread.java:764)
E/AndroidRuntime(10153): FATAL EXCEPTION: AsyncTask #2
E/AndroidRuntime(10153): Process: br.com.emdurb.talonarioeletronico, PID: 10153
E/AndroidRuntime(10153): java.lang.IllegalArgumentException: Unsupported value: java.io.IOException: read failed, socket might closed or timeout, read ret: -1
E/AndroidRuntime(10153): 	at io.flutter.plugin.common.StandardMessageCodec.writeValue(StandardMessageCodec.java:293)
E/AndroidRuntime(10153): 	at io.flutter.plugin.common.StandardMethodCodec.encodeErrorEnvelope(StandardMethodCodec.java:70)
E/AndroidRuntime(10153): 	at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler$1.error(MethodChannel.java:208)
E/AndroidRuntime(10153): 	at io.github.edufolly.flutterbluetoothserial.FlutterBluetoothSerialPlugin.lambda$connect$0(FlutterBluetoothSerialPlugin.java:229)
E/AndroidRuntime(10153): 	at io.github.edufolly.flutterbluetoothserial.-$$Lambda$FlutterBluetoothSerialPlugin$52C5RUgoP_r70sgAnP8ZXC1Ysz0.run(Unknown Source:6)
E/AndroidRuntime(10153): 	at android.os.AsyncTask$SerialExecutor$1.run(Unknown Source:2)
E/AndroidRuntime(10153): 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
E/AndroidRuntime(10153): 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
E/AndroidRuntime(10153): 	at java.lang.Thread.run(Thread.java:764)
E/MQSEventManagerDelegate(10153): failed to get MQSService.
I/Process (10153): Sending signal. PID: 10153 SIG: 9

edufolly, 2018-12-16:
Hi @rafaelterada,
I'll try to simulate this error, but it'll be easier for me to see what you are doing. If you want, send me an email to schedule a time to talk about.

edufolly, 2018-12-20:
Hi @rafaelterada,

I published a new version of the plugin with your PRs. Please, make some tests and if you find some problem open a new issue.

Best.

closed 2018-12-20 by edufolly