edufolly/flutter_bluetooth_serial #12: Message other than String Type

kidzen, 2018-12-22:


edufolly, 2018-12-23:
Hi @kidzen,

I believe that is not difficult but at this moment I don't have the environment to make tests.

```dart
import 'dart:typed_data';
class FlutterBluetoothSerial {
  ...
  Future<dynamic> writeBytes(Uint8List message)  =>
      _channel.invokeMethod('writeBytes', {'message': message});
  ...
}
```

```java
public class FlutterBluetoothSerialPlugin ...
  ...
  public void onMethodCall ... {
    ...
            case "writeBytes":
                if (arguments.containsKey("message")) {
                    byte[] message = (byte[]) arguments.get("message");
                    write(result, message);
                } else {
                    result.error("invalid_argument", "argument 'message' not found", null);
                }
                break;
   ...
  }
  ...
    private void write(Result result, byte[] message) {
        if (THREAD == null) {
            result.error("write_error", "not connected", null);
            return;
        }
        try {
            THREAD.write(message);
            result.success(true);
        } catch (Exception ex) {
            Log.e(TAG, ex.getMessage(), ex);
            result.error("write_error", ex.getMessage(), exceptionToString(ex));
        }
    }

    private void write(Result result, String message) {
        write(result, message.getBytes());
    }
  ...
}
```
We can use this example as base code. ;-)

Maybe @rafaelterada could help us with tests and PR.

Best.

kidzen, 2018-12-24:
tq @edufolly  for your fast response...i will try it out and post the result back to u...tq so much!

rafaelterada, 2018-12-27:
Hi, I made the changes, however I can not test passing byte []. Can you do the tests and not give a return?
Test with this:
https://github.com/rafaelterada/flutter_bluetooth_serial/tree/patch-1

edufolly, 2019-02-05:
Hi @rafaelterada, 

could you provide a PR with these features?

edufolly, 2019-03-18:
Hi folks,

PR was merged.

Thanks.

closed 2019-03-18 by edufolly