edufolly/flutter_bluetooth_serial #14: Invoke-customs are only supported starting with android 0 --min-api 26

jingramwright, 2019-01-13:
I was getting the issue outlined in the title when attempting to run the example included in the repo.

Googling the error message pointed me in the direction of this Stack Overflow thread:
[https://stackoverflow.com/questions/49891730/invoke-customs-are-only-supported-starting-with-android-0-min-api-26](https://stackoverflow.com/questions/49891730/invoke-customs-are-only-supported-starting-with-android-0-min-api-26)

I had to add the following to the build.gradle file within "android > app".

compileOptions {
    sourceCompatibility JavaVersion.VERSION_1_8
    targetCompatibility JavaVersion.VERSION_1_8
}

Thought it might be worth adding to the readme in case anyone else encounters this issue.

Thanks!

JMRjose, 2019-01-16:
yes this works thanks.

closed 2019-03-19 by edufolly

yuanchen1, 2019-04-08:
如果还有错误的话　请在gradle.properties 加入　
android.useAndroidX=true
android.enableJetifier=true