edufolly/flutter_bluetooth_serial #17: Communication between 2 phones is not sent through

fmatosqg, 2019-03-01:
Hi,

I picked up 2 phones and after pairing them I can connect one to another. This is confirmed on the receiving phone because the "connect" button automatically changes to "disconnect".

However after a few seconds both buttons return to "connect", followed by a message on console.

> D/FlutterBluePlugin(14133): android.bluetooth.device.action.ACL_DISCONNECTED
> D/FlutterBluePlugin(14133): android.bluetooth.device.action.ACL_CONNECTED
> D/FlutterBluePlugin(14133): android.bluetooth.device.action.ACL_DISCONNECTED
> D/FlutterBluePlugin(14133): android.bluetooth.device.action.ACL_CONNECTED
> D/FlutterBluePlugin(14133): android.bluetooth.device.action.ACL_DISCONNECTED
> D/FlutterBluePlugin(14133): android.bluetooth.device.action.ACL_CONNECTED
> D/FlutterBluePlugin(14133): android.bluetooth.device.action.ACL_DISCONNECTED
> 

I also tried typing something and pressing send, but I couldn't make the message show up on the other phone.

PsychoXIVI, 2019-05-30:
The problem is the socket can't be created, since they are connecting, but no one is listening. 

There must be listening/server mode implemented (also in platform code). In #35 I am working on it.

SubasreeArvind, 2020-07-14:
Hi,
I too want to connect two android mobiles and enable chat feature amongst them using classic bluetooth however as mentioned, there isn't any listener side code in the example program. 
@PsychoXIVI has there been any development/solution to this issue?

staminna, 2021-09-21:
I also looking for this solution. @SubasreeArvind Did you manage to find out how to do this?