edufolly/flutter_bluetooth_serial #23: Is it possible to print Qr Image by sending bytes?

geoffreylzf, 2019-03-29:


geoffreylzf, 2019-03-29:
I manage to send Uint8List by following code

```
Future<Uint8List> _getQrByte() async {
    RenderRepaintBoundary boundary =
        globalKey.currentContext.findRenderObject();
    var image = await boundary.toImage();
    var byteData = await image.toByteData();
    return byteData.buffer.asUint8List();
  }
```
In another function I call
`await _bluetooth.writeBytes(bytes);`

But the printer out random which is incorrect and very long.

Is there any mistakes on my code?

kidzen, 2019-04-17:
does your qr come from a real image and converted to byte for transfer?if it is, then its surely fail..
as far as i know, image transfer through byte loses some of its information... i myself end up transfered through path..

PsychoXIVI, 2019-06-18:
I think it is printer specific problem. The `flutter_bluetooth_serial` library just allow to communicate through Bluetooth. You should look for what Bluetooth protocol is used and how the commands should be formed.

closed 2019-06-27 by PsychoXIVI