edufolly/flutter_bluetooth_serial #25: Closing the app leaks an IntentReceiver because of onListen

nanobug, 2019-04-25:
Maybe there should be some 'shutdown' api to make a unregister call for the stateStreamHandler?

PsychoXIVI, 2019-06-21:
As far as I know Flutter does not have API for disposing platform code on app termination. Problem is, there is no way to use `onDestroy` method of `Activity` (https://github.com/flutter/flutter/issues/21982). 

There is possibility to detect application life-cycle, including suspending and resuming - which potentially could be used in future 🤔 . I think it would cause `onStateChanged` and similar streams to be closed every time app goes suspended, so user will need to reopen the stream(s).

PsychoXIVI, 2019-06-28:
Flutter plugin registry exposes [`addViewDestroyListener` ](https://api.flutter.dev/javadoc/io/flutter/plugin/common/PluginRegistry.Registrar.html#addViewDestroyListener-io.flutter.plugin.common.PluginRegistry.ViewDestroyListener-) which allows the library take a part in `onDestroy` of the application... I think I was blind, I did not see it...

Commit with fix landing in few minutes on `develop` branch, so closing for now.

closed 2019-06-28 by PsychoXIVI