edufolly/flutter_bluetooth_serial #27: Trying to connect to a turned off device gives a exception

joaowd, 2019-05-06:
If I try to connect to a turned off device I get the following error:

D/HWUIExtension(18400): MTKProgramCache.generateProgram: 562984313159681
D/HWUIExtension(18400): createProgram 0x0002000800000001, binary 0x95f21698, length 11759, format 37168 within 1848ns
W/ResourcesManager(18400): key.mResDir: /data/app/com.example.nexso-WbEriBSNsstF1_gci-xIvg==/base.apk, ignore idmapPath: /system/overlay/framework-res-overlay.apk
W/BluetoothAdapter(18400): getBluetoothService() called with no BluetoothManagerCallback
E/FlutterBluePlugin(18400): read failed, socket might closed or timeout, read ret: -1
E/FlutterBluePlugin(18400): java.io.IOException: read failed, socket might closed or timeout, read ret: -1
E/FlutterBluePlugin(18400): 	at android.bluetooth.BluetoothSocket.readAll(BluetoothSocket.java:685)
E/FlutterBluePlugin(18400): 	at android.bluetooth.BluetoothSocket.readInt(BluetoothSocket.java:697)
E/FlutterBluePlugin(18400): 	at android.bluetooth.BluetoothSocket.connect(BluetoothSocket.java:374)
E/FlutterBluePlugin(18400): 	at io.github.edufolly.flutterbluetoothserial.FlutterBluetoothSerialPlugin.lambda$connect$0$FlutterBluetoothSerialPlugin(FlutterBluetoothSerialPlugin.java:248)
E/FlutterBluePlugin(18400): 	at io.github.edufolly.flutterbluetoothserial.-$$Lambda$FlutterBluetoothSerialPlugin$52C5RUgoP_r70sgAnP8ZXC1Ysz0.run(Unknown Source:6)
E/FlutterBluePlugin(18400): 	at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:245)
E/FlutterBluePlugin(18400): 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1162)
E/FlutterBluePlugin(18400): 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:636)
E/FlutterBluePlugin(18400): 	at java.lang.Thread.run(Thread.java:764)

Isti01, 2019-05-13:
The problem is that when this happens, it fails silently, and the connect method never returns.

PsychoXIVI, 2019-06-18:
In #35 the issue seems to not exist - every failed connection is thrown either. I imagine there was no error pushed to common code (Dart) from platform code (Java) like `result.error(...)`. 
For now I will close the issue, since #35 is right around the corner.

closed 2019-06-18 by PsychoXIVI