edufolly/flutter_bluetooth_serial #30: How can I check Bluetooth On/Off State?

JAICHANGPARK, 2019-05-28:
I'd like to check Bluetooth On/Off State before connection 

Can I check this?

FWeissenb, 2019-05-29:
There is an "isOn" 

Should be something like (untested)

```
var isOn = await FlutterBluetoothSerial.instance.isOn;

if(isOn) {
//TODO
}
````

PsychoXIVI, 2019-05-30:
In few days there might be: See #35

PsychoXIVI, 2019-06-18:
The pull request #35 will be merged soon, and it adds the option, so the issue is being closed. 
And there is syntax:
```Dart
// Single call check
bool isEnabled = await FlutterBluetoothSerial.instance.isEnabled;

// Or listening for changes
FlutterBluetoothSerial.instance.onStateChanged().listen((BluetoothState state) {
    // The `BluetoothState` has enum-like interface with `isEnabled` getter (true if `state == BluetoothState.STATE_ON` 
    print(state.isEnabled); 
});
```

closed 2019-06-18 by PsychoXIVI