edufolly/flutter_bluetooth_serial #36: App Crashes on connection

droidchef, 2019-06-02:
When I try to connect to a paired HC-05 Module running on my Arduino. The example app just crashes with this stack trace

```
W/BluetoothAdapter(24727): getBluetoothService() called with no BluetoothManagerCallback
D/FlutterBluePlugin(24727): android.bluetooth.device.action.ACL_CONNECTED
E/FlutterBluePlugin(24727): Methods marked with @UiThread must be executed on the main thread. Current thread: AsyncTask #2
E/FlutterBluePlugin(24727): java.lang.RuntimeException: Methods marked with @UiThread must be executed on the main thread. Current thread: AsyncTask #2
E/FlutterBluePlugin(24727): 	at io.flutter.embedding.engine.FlutterJNI.ensureRunningOnMainThread(FlutterJNI.java:605)
E/FlutterBluePlugin(24727): 	at io.flutter.embedding.engine.FlutterJNI.invokePlatformMessageResponseCallback(FlutterJNI.java:556)
E/FlutterBluePlugin(24727): 	at io.flutter.embedding.engine.dart.DartMessenger$Reply.reply(DartMessenger.java:131)
E/FlutterBluePlugin(24727): 	at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler$1.success(MethodChannel.java:225)
E/FlutterBluePlugin(24727): 	at io.github.edufolly.flutterbluetoothserial.FlutterBluetoothSerialPlugin.lambda$connect$0(FlutterBluetoothSerialPlugin.java:257)
E/FlutterBluePlugin(24727): 	at io.github.edufolly.flutterbluetoothserial.-$$Lambda$FlutterBluetoothSerialPlugin$52C5RUgoP_r70sgAnP8ZXC1Ysz0.run(Unknown Source:6)
E/FlutterBluePlugin(24727): 	at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:245)
E/FlutterBluePlugin(24727): 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
E/FlutterBluePlugin(24727): 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
E/FlutterBluePlugin(24727): 	at java.lang.Thread.run(Thread.java:764)
E/AndroidRuntime(24727): FATAL EXCEPTION: AsyncTask #2
E/AndroidRuntime(24727): Process: com.example.timecube, PID: 24727
E/AndroidRuntime(24727): java.lang.IllegalStateException: Reply already submitted
E/AndroidRuntime(24727): 	at io.flutter.embedding.engine.dart.DartMessenger$Reply.reply(DartMessenger.java:126)
E/AndroidRuntime(24727): 	at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler$1.error(MethodChannel.java:230)
E/AndroidRuntime(24727): 	at io.github.edufolly.flutterbluetoothserial.FlutterBluetoothSerialPlugin.lambda$connect$0(FlutterBluetoothSerialPlugin.java:260)
E/AndroidRuntime(24727): 	at io.github.edufolly.flutterbluetoothserial.-$$Lambda$FlutterBluetoothSerialPlugin$52C5RUgoP_r70sgAnP8ZXC1Ysz0.run(Unknown Source:6)
E/AndroidRuntime(24727): 	at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:245)
E/AndroidRuntime(24727): 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
E/AndroidRuntime(24727): 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
E/AndroidRuntime(24727): 	at java.lang.Thread.run(Thread.java:764)
D/OSTracker(24727): OS Event: crash
D/AbstractTracker(24727): Event success
I/Process (24727): Sending signal. PID: 24727 SIG: 9
Lost connection to device.
```

PsychoXIVI, 2019-06-03:
This is related to the new versions of Flutter, sorry. I fixed it already at #35, but if you need fix immediately for current version of the `flutter_bluetooth_serial` library, you can modify platform code (Java) to run all `EventSink`/`EventChannel` related things (`success`/`error`/`endOfStream`) on `UiThread`.

droidchef, 2019-06-04:
@PsychoXIVI Thank you for the response. Your fix indeed works. If it is going to take you a while to finish your PR, do you mind if I contribute this fix back to the project? I don't want to be using the plugin from my work and then having to make changes in future 😛

PsychoXIVI, 2019-06-05:
@droidchef Sure thing, you can add them to this version too. 

You can also consider using my version of the plugin for now if you want:
+ either download it and point locally in `pubspec.yaml` (something like `flutter_bluetooth_serial:` (\n) `  path: ../../btplugin`) 
+ or point remotely in `pubspec.yaml` to some of commits on my fork of the plugin (see https://flutter.dev/docs/development/packages-and-plugins/using-packages#dependencies-on-unpublished-packages).

droidchef, 2019-06-06:
@PsychoXIVI I am also working with Arduino HC-05 module with this plugin 😄

ianhzhang, 2019-06-09:
I pull the full code from github from the github.   Run the example file.   cd example.  flutter run.  When I tried to connect to my ubuntu machine, I get following error.  Similar, not the same.

D/AwareBitmapCacher(29903): handleInit switch not opened pid=29903
V/AudioManager(29903): playSoundEffect   effectType: 0
V/AudioManager(29903): querySoundEffectsEnabled...
V/AudioManager(29903): playSoundEffect   effectType: 0
V/AudioManager(29903): querySoundEffectsEnabled...
D/BluetoothSocket(29903): BT connect calling pid/uid = 29903/10025
W/BluetoothAdapter(29903): getBluetoothService() called with no BluetoothManagerCallback
E/flutter (29903): [ERROR:flutter/lib/ui/ui_dart_state.cc(148)] Unhandled Exception: NoSuchMethodError: The method 'catchError' was called on null.
E/flutter (29903): Receiver: null
E/flutter (29903): Tried calling: catchError(Closure: (dynamic) => Null)
E/flutter (29903): #0      Object.noSuchMethod (dart:core-patch/object_patch.dart:50:5)
E/flutter (29903): #1      _MyAppState._connect.<anonymous closure> (package:flutter_bluetooth_serial_example/main.dart:197:38)
E/flutter (29903): #2      _rootRunUnary (dart:async/zone.dart:1132:38)
E/flutter (29903): #3      _CustomZone.runUnary (dart:async/zone.dart:1029:19)
E/flutter (29903): #4      _FutureListener.handleValue (dart:async/future_impl.dart:126:18)
E/flutter (29903): #5      Future._propagateToListeners.handleValueCallback (dart:async/future_impl.dart:639:45)
E/flutter (29903): #6      Future._propagateToListeners (dart:async/future_impl.dart:668:32)
E/flutter (29903): #7      Future._complete (dart:async/future_impl.dart:473:7)
E/flutter (29903): #8      _SyncCompleter.complete (dart:async/future_impl.dart:51:12)
E/flutter (29903): #9      _AsyncAwaitCompleter.complete (dart:async-patch/async_patch.dart:28:18)
E/flutter (29903): #10     _completeOnAsyncReturn (dart:async-patch/async_patch.dart:294:13)
E/flutter (29903): #11     FlutterBluetoothSerial.isConnected (package:flutter_bluetooth_serial/flutter_bluetooth_serial.dart)
E/flutter (29903): #12     _asyncThenWrapperHelper.<anonymous closure> (dart:async-patch/async_patch.dart:77:64)
E/flutter (29903): #13     _rootRunUnary (dart:async/zone.dart:1132:38)
E/flutter (29903): #14     _CustomZone.runUnary (dart:async/zone.dart:1029:19)
E/flutter (29903): #15     _FutureListener.handleValue (dart:async/future_impl.dart:126:18)
E/flutter (29903): #16     Future._propagateToListeners.handleValueCallback (dart:async/future_impl.dart:639:45)
E/flutter (29903): #17     Future._propagateToListeners (dart:async/future_impl.dart:668:32)
E/flutter (29903): #18     Future._complete (dart:async/future_impl.dart:473:7)
E/flutter (29903): #19     _SyncCompleter.complete (dart:async/future_impl.dart:51:12)
E/flutter (29903): #20     _AsyncAwaitCompleter.complete (dart:async-patch/async_patch.dart:28:18)
E/flutter (29903): #21     _completeOnAsyncReturn (dart:async-patch/async_patch.dart:294:13)
E/flutter (29903): #22     MethodChannel.invokeMethod (package:flutter/src/services/platform_channel.dart)
E/flutter (29903): #23     _asyncThenWrapperHelper.<anonymous closure> (dart:async-patch/async_patch.dart:77:64)
E/flutter (29903): #24     _rootRunUnary (dart:async/zone.dart:1132:38)
E/flutter (29903): #25     _CustomZone.runUnary (dart:async/zone.dart:1029:19)
E/flutter (29903): #26     _FutureListener.handleValue (dart:async/future_impl.dart:126:18)
E/flutter (29903): #27     Future._propagateToListeners.handleValueCallback (dart:async/future_impl.dart:639:45)
E/flutter (29903): #28     Future._propagateToListeners (dart:async/future_impl.dart:668:32)
E/flutter (29903): #29     Future._completeWithValue (dart:async/future_impl.dart:483:5)
E/flutter (29903): #30     Future._asyncComplete.<anonymous closure> (dart:async/future_impl.dart:513:7)
E/flutter (29903): #31     _rootRun (dart:async/zone.dart:1124:13)
E/flutter (29903): #32     _CustomZone.run (dart:async/zone.dart:1021:19)
E/flutter (29903): #33     _CustomZone.runGuarded (dart:async/zone.dart:923:7)
E/flutter (29903): #34     _CustomZone.bindCallbackGuarded.<anonymous closure> (dart:async/zone.dart:963:23)
E/flutter (29903): #35     _microtaskLoop (dart:async/schedule_microtask.dart:41:21)
E/flutter (29903): #36     _startMicrotaskLoop (dart:async/schedule_microtask.dart:50:5)
E/flutter (29903): 
D/FlutterBluePlugin(29903): android.bluetooth.device.action.ACL_CONNECTED
E/FlutterBluePlugin(29903): read failed, socket might closed or timeout, read ret: -1
E/FlutterBluePlugin(29903): java.io.IOException: read failed, socket might closed or timeout, read ret: -1
E/FlutterBluePlugin(29903): 	at android.bluetooth.BluetoothSocket.readAll(BluetoothSocket.java:793)
E/FlutterBluePlugin(29903): 	at android.bluetooth.BluetoothSocket.readInt(BluetoothSocket.java:807)
E/FlutterBluePlugin(29903): 	at android.bluetooth.BluetoothSocket.connect(BluetoothSocket.java:413)
E/FlutterBluePlugin(29903): 	at io.github.edufolly.flutterbluetoothserial.FlutterBluetoothSerialPlugin.lambda$connect$0(FlutterBluetoothSerialPlugin.java:319)
E/FlutterBluePlugin(29903): 	at io.github.edufolly.flutterbluetoothserial.-$$Lambda$FlutterBluetoothSerialPlugin$52C5RUgoP_r70sgAnP8ZXC1Ysz0.run(Unknown Source:6)
E/FlutterBluePlugin(29903): 	at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:246)
E/FlutterBluePlugin(29903): 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
[... cut ...]
E/flutter (29903): #25     _WidgetsFlutterBinding&BindingBase&GestureBinding._handlePointerDataPacket (package:flutter/src/gestures/binding.dart:86:7)
E/flutter (29903): #26     _rootRunUnary (dart:async/zone.dart:1136:13)
E/flutter (29903): #27     _CustomZone.runUnary (dart:async/zone.dart:1029:19)
E/flutter (29903): #28     _CustomZone.runUnaryGuarded (dart:async/zone.dart:931:7)
E/flutter (29903): #29     _invoke1 (dart:ui/hooks.dart:233:10)
E/flutter (29903): #30     _dispatchPointerDataPacket (dart:ui/hooks.dart:154:5)
E/flutter (29903): 
D/FlutterBluePlugin(29903): android.bluetooth.device.action.ACL_DISCONNECTED

PsychoXIVI, 2019-06-09:
@ianhzhang , issue from your is different kind problem: `read failed, socket might closed or timeout, read ret: -1`.

By the way, you REALLY should post the logs on some gist (or pastebin) or at least by using markdown (code snippet):
[image]

Please have it on mind next time posting code or logs.

Anyways, the #18 address this problem, so look here (https://github.com/edufolly/flutter_bluetooth_serial/issues/18)

JAICHANGPARK, 2019-06-17:
I HAVE TO SAME ERROR 

>Activity com.flutter_app.MainActivity has leaked IntentReceiver io.github.edufolly.flutterbluetoothserial.FlutterBluetoothSerialPlugin$2$1@152050f that was originally registered here. Are you missing a call to unregisterReceiver()?

>.MainActivity has leaked IntentReceiver io.github.edufolly.flutterbluetoothserial.FlutterBluetoothSerialPlugin$2$1@152050f that was originally registered here. Are you missing a call to unregisterReceiver()?

HOW TO FIX IT?

JAICHANGPARK, 2019-06-17:
`E/FlutterBluePlugin( 9642): Methods marked with @UiThread must be executed on the main thread. Current thread: AsyncTask #2
E/FlutterBluePlugin( 9642): java.lang.RuntimeException: Methods marked with @UiThread must be executed on the main thread. Current thread: AsyncTask #2
E/FlutterBluePlugin( 9642): 	at io.flutter.embedding.engine.FlutterJNI.ensureRunningOnMainThread(FlutterJNI.java:605)
E/FlutterBluePlugin( 9642): 	at io.flutter.embedding.engine.FlutterJNI.invokePlatformMessageResponseCallback(FlutterJNI.java:556)
E/FlutterBluePlugin( 9642): 	at io.flutter.embedding.engine.dart.DartMessenger$Reply.reply(DartMessenger.java:131)
E/FlutterBluePlugin( 9642): 	at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler$1.success(MethodChannel.java:225)
E/FlutterBluePlugin( 9642): 	at io.github.edufolly.flutterbluetoothserial.FlutterBluetoothSerialPlugin.lambda$connect$0(FlutterBluetoothSerialPlugin.java:323)
E/FlutterBluePlugin( 9642): 	at io.github.edufolly.flutterbluetoothserial.-$$Lambda$FlutterBluetoothSerialPlugin$52C5RUgoP_r70sgAnP8ZXC1Ysz0.run(Unknown Source:6)
E/FlutterBluePlugin( 9642): 	at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:245)
E/FlutterBluePlugin( 9642): 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1162)
E/FlutterBluePlugin( 9642): 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:636)
E/FlutterBluePlugin( 9642): 	at java.lang.Thread.run(Thread.java:764)
E/FlutterBluePlugin( 9642): Reply already submitted
E/FlutterBluePlugin( 9642): java.lang.IllegalStateException: Reply already submitted
E/FlutterBluePlugin( 9642): 	at io.flutter.embedding.engine.dart.DartMessenger$Reply.reply(DartMessenger.java:126)
E/FlutterBluePlugin( 9642): 	at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler$1.error(MethodChannel.java:230)
E/FlutterBluePlugin( 9642): 	at io.github.edufolly.flutterbluetoothserial.FlutterBluetoothSerialPlugin.lambda$connect$0(FlutterBluetoothSerialPlugin.java:326)
E/FlutterBluePlugin( 9642): 	at io.github.edufolly.flutterbluetoothserial.-$$Lambda$FlutterBluetoothSerialPlugin$52C5RUgoP_r70sgAnP8ZXC1Ysz0.run(Unknown Source:6)
E/FlutterBluePlugin( 9642): 	at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:245)
E/FlutterBluePlugin( 9642): 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1162)
E/FlutterBluePlugin( 9642): 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:636)
E/FlutterBluePlugin( 9642): 	at java.lang.Thread.run(Thread.java:764)
E/AndroidRuntime( 9642): FATAL EXCEPTION: AsyncTask #2
E/AndroidRuntime( 9642): Process: com.angelrobotics.flutter_angel_suit_app, PID: 9642
E/AndroidRuntime( 9642): java.lang.IllegalStateException: Reply already submitted
E/AndroidRuntime( 9642): 	at io.flutter.embedding.engine.dart.DartMessenger$Reply.reply(DartMessenger.java:126)
E/AndroidRuntime( 9642): 	at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler$1.error(MethodChannel.java:230)
E/AndroidRuntime( 9642): 	at io.github.edufolly.flutterbluetoothserial.FlutterBluetoothSerialPlugin.lambda$connect$0(FlutterBluetoothSerialPlugin.java:330)
E/AndroidRuntime( 9642): 	at io.github.edufolly.flutterbluetoothserial.-$$Lambda$FlutterBluetoothSerialPlugin$52C5RUgoP_r70sgAnP8ZXC1Ysz0.run(Unknown Source:6)
E/AndroidRuntime( 9642): 	at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:245)
E/AndroidRuntime( 9642): 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1162)
E/AndroidRuntime( 9642): 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:636)
E/AndroidRuntime( 9642): 	at java.lang.Thread.run(Thread.java:764)
`

PsychoXIVI, 2019-06-17:
@JAICHANGPARK Two solutions have been already mentioned (at least for `UiThread` issue).

And for the `IntentReceiver` - there is no much to do for now. The receiver class isn't unregistered in current code. Well, it isn't critical failure, but is is bad to left such things undisposed to Android system.

JAICHANGPARK, 2019-06-18:
@PsychoXIVI  Thanks for comment :) 
I checked your PR code .. How to check bluetooth connection, disconnection state in your code?

PsychoXIVI, 2019-06-18:
@JAICHANGPARK The connection means that `connection.output` and `connection.input` streams (`Stream<Uint8List>`) are open. 

Then if the connection is closed by remote host, or just lack of connectivity, you can detect it by `onDone` of `input`.

If you are (the device) who are closing the connection, the `onDone` callback will not be called (this allow to detect who is disconnecting). If you want to have universal callback on disconnection (regardless who caused the disconnection) you can create some `Function` field in your environment and then pass it to `onDone` of `input` and also call it manually when disconnecting by you (after `connection.finish()`).

The main difference between old version (0.0.5) and my version is fact, there might be multiple connections at the same time. Observing `BluetoothStatus` is just information about the adapter status itself (off/turning on/on/turning off). You need to create `BluetoothConnection` instance. 

```Dart
try {
    BluetoothConnection connection = await BluetoothConnection.toAddress(address);
    print('Connected to the device');
    connection.input.listen((Uint8List data) {
        print('Data incoming: ${ascii.decode(data)}');
        connection.output.add(data); // Sending data
        if (ascii.decode(data).contains('!')) {
            connection.finish(); // Closing connection
            print('Disconnecting by local host');
        }
    }).onDone(() {
        print('Disconnected by remote request');
    });
}
catch (exception) {
    print('Cannot connect, exception occured');
}
```

PsychoXIVI, 2019-06-18:
The issue is about the `@UiThread` problem in newer versions of Flutter. It is solved by #35 and the #35 PR is coming very soon so I am closing the issue for now.

closed 2019-06-18 by PsychoXIVI

Amin-Matola, 2020-09-02:
> @JAICHANGPARK The connection means that `connection.output` and `connection.input` streams (`Stream<Uint8List>`) are open.
>
> Then if the connection is closed by remote host, or just lack of [...]

I use this exact approach to connect to a device, but all I get is the **catch** block's message... what would be the issue then, and how can I possibly fix the exception?
Here is my current code:

```
BluetoothConnection.toAddress(address)
      .then((conn){
                  print("Connected to device");
        })
      .catchError((error){
          showDialog(context: context, builder: (d){
            return AlertDialog(
              title: Text("Connection Problem"),
              content: Text("There was an error connecting to device $address"),
              actions: [
                FlatButton(
                  child: Text("Ok"),
                  onPressed: (){
                    Navigator.pop(context);
                  },
                )
              ],
            );
          });
        });
```