edufolly/flutter_bluetooth_serial #39: Formating bytes coming from onRead or Input and data always coming with "\r\n"

chrisgate, 2019-06-21:
Below console information is coming from RFID Reader and it can read upto 500 of it at the same time but i noticed it always come with \r\n at the back when step over in debugging e.g EP: E2000015151001591070AC5F\r\n EP: E2007B037A3C4DF16D7AB537
and below when is the console messages:
I/flutter (10731): CS: .iv
I/flutter (10731): CS: .iv
I/flutter (10731): EP: 300833B2DDD9014000000000
I/flutter (10731): EP: E2000015151001591070AC5F
I/flutter (10731): EP: E2007B037A3C4DF16D7AB537
I/flutter (10731): EP: E2000015151001581070AC57
I/flutter (10731): EP: E2000015151002501200A272
I/flutter (10731): CS: .iv
I/flutter (10731): EP: 300833B2DDD9014000000000
I/flutter (10731): EP: E2000015151001591070AC5F
I/flutter (10731): EP: E2007B037A3C4DF16D7AB537
I/flutter (10731): EP: E2000015151001581070AC57
I/flutter (10731): EP: E2000015151002501200A272
I/flutter (10731): EP: E2000015151001271070AC1F
I/flutter (10731): EP: E2000015151001821060AEA7
I/flutter (10731): EP: E2000015151001580810C89C
I/flutter (10731): EP: 300833B2DDD9014000000000
I/flutter (10731): OK:

Thank You

PsychoXIVI, 2019-06-21:
Could you post code used to receive the data?

_I assuming you are using 0.1.0 version, point me, if I am wrong._

chrisgate, 2019-06-22:
Thanks for you and yes i am 0.1.0 
and the code below from the example :https://github.com/edufolly/flutter_bluetooth_serial/blob/master/example/lib/ChatPage.dart

 void _onDataReceived(Uint8List data) {
    // Allocate buffer for parsed data
    int backspacesCounter = 0;
    data.forEach((byte) {
      if (byte == 8 || byte == 127) {
        backspacesCounter++;
      }
    });
    Uint8List buffer = Uint8List(data.length - backspacesCounter);
    int bufferIndex = buffer.length;

    // Apply backspace control character
    backspacesCounter = 0;
    for (int i = data.length - 1; i >= 0; i--) {
      if (data[i] == 8 || data[i] == 127) {
        backspacesCounter++;
      }
      else {
        if (backspacesCounter > 0) {
          backspacesCounter--;
        }
        else {
          buffer[--bufferIndex] = data[i];
        }
      }
    }

    // Create message if there is new line character
    String dataString = String.fromCharCodes(buffer);
    int index = buffer.indexOf(13);
    if (~index != 0) { // \r\n
      setState(() {
        messages.add(_Message(1, 
          backspacesCounter > 0 
            ? _messageBuffer.substring(0, _messageBuffer.length - backspacesCounter) 
            : _messageBuffer
          + dataString.substring(0, index)
        ));
        _messageBuffer = dataString.substring(index);
      });
    }
    else {
      _messageBuffer = (
        backspacesCounter > 0 
          ? _messageBuffer.substring(0, _messageBuffer.length - backspacesCounter) 
          : _messageBuffer
        + dataString
      );
    }
  }

I use below to console:
 Container(
            child: Text((text) {
              **String a = _message.text.trim();
              print(a);**
              return text == '/shrug' ? '¯\\_(ツ)_/¯' : text;
            } (_message.text.trim()), style: TextStyle(color: Colors.white)),
            padding: EdgeInsets.all(12.0),
            margin: EdgeInsets.only(bottom: 8.0, left: 8.0, right: 8.0),
            width: 222.0,
            decoration: BoxDecoration(color: _message.whom == clientID ? Colors.blueAccent : Colors.grey, borderRadius: BorderRadius.circular(7.0)),
          ),

Thanks

chrisgate, 2019-06-22:
and it possible to cast is to some kind of model?

PsychoXIVI, 2019-06-22:
In the example there were simple buffering and backspace handling created to make it work as chat with simple chat server ([with Raspberry Pi, `picocom`, ...](https://github.com/edufolly/flutter_bluetooth_serial/blob/master/example/README.md#chat-example)). I don't think it should be used in every case...

### Is it possible to cast is to some kind of model since i know what coming in?

Yes, definitely. In your case you might want to write some Stream parser. Please take a look at [`BackgroundCollectingTask` example code](https://github.com/edufolly/flutter_bluetooth_serial/blob/master/example/lib/BackgroundCollectingTask.dart). The data incoming are buffered and parsed into `DataSample`s objects.
```Dart
    _connection.input.listen((data) {
      _buffer += data;

      while (true) {
        // If there is a sample, and it is full sent
        int index = _buffer.indexOf('t'.codeUnitAt(0));
        if (index >= 0 && _buffer.length - index >= 7) {
          final DataSample sample = DataSample(
            temperature1: (_buffer[index + 1] + _buffer[index + 2] / 100), // This is example specific parsing, 
            temperature2: (_buffer[index + 3] + _buffer[index + 4] / 100), // just simplest (non-efficient) decoding doubles 
            waterpHlevel: (_buffer[index + 5] + _buffer[index + 6] / 100),
            timestamp: DateTime.now()
          );
          _buffer.removeRange(0, index + 7);

          samples.add(sample);
          notifyListeners();
          //print("${sample.timestamp.toString()} -> ${sample.temperature1} / ${sample.temperature2}");
        }
        // Otherwise break
        else {
          break;
        }
      }
    }).onDone(() {
      inProgress = false;
      notifyListeners();
    });
```
If you know structure/layout of your data, you could create such stream parser for it and then add the parser objects to another stream (true good Dart way I think) or store the objects at some list (like in the example).

### Is it only `byte[]` that can come from `BluetoothSocket` at the native side?

`BluetoothSocket` is indeed passing just bytes. The user of the socket decides what do with the bytes - the same approach is used in the library. You can work with the raw bytes stream or you can use some functions like `ascii.decode` and `ascii.encode` to send/receive strings. You should also consider buffering.

### Future...

The library lacks some simpler utility to make it easy to adopt any Bluetooth serial communication into modeled request/response communication. There might be atomic access to such utility and other things to make it work smoothly. I am thinking of something like:
```Dart
C.request("getStatus", {"someArg": 123}).onResponse((String r) {
  MyStatus status = MyStatus.fromString(r);
  //...
});
```
It is on `To Do` list already.

chrisgate, 2019-06-23:
Thanks ............I will look into BackgroundCollectingTask example code and revert .......Great work

PsychoXIVI, 2019-06-23:
Closing the issue for now. If you need more information or help at this problem, write here (issue will be reopened).

closed 2019-06-23 by PsychoXIVI

PsychoXIVI, 2019-06-29:
There was #41 opened as discussion about the utilities for easing data transmission I mentioned above.