edufolly/flutter_bluetooth_serial #41: Utilities for request-response connection manner

PsychoXIVI, 2019-06-29:
Here is issue to discuss form and expected syntax for the utilities which will be targeted in next milestone.

General propose for the utilities: **ease communication between your Bluetooth devices over RFCOMM (serial) protocol**.

Assumptions:
* whole utilities must be optional, leaving raw byte streams available,
* should be consuming raw streams or providing separate interface on `BluetoothConnection`,

PsychoXIVI, 2019-06-29:
My current proposal:
* packet system which manages buffering before handling single package,
* data parsing by user (`fromBytes`/`toBytes` could be good for user data models),
* there will be normal packets helper and ensured one (checksum with anti-corruption via retransmission) 

Some sample code would look like:
```Dart
//BluetoothConnectionAsPackets packets = connection.asPackets; // and `connection.asPackets.receive(...);`
BluetoothConnectionAsPackets packets = BluetoothConnectionAsPackets.fromConnection(connection);
packets.receive((int type, Uint8List data) {
  switch (type) {
    case 0x01: // These could be also closed into enum-like class.
      dataSamplesStream.add(DataSample.fromBytes(data));
      break;
    case 0x02:
      // Deprecated types IDs should not be reused in real environment
      break;
    case 0x03:
      packets.send(0x04, data); // or `connection.asPackets.send(0x04, data);`
      break;
    case 0x04:
      packets.send(0x11, someDataObject.toBytes())
      break;
    //...
    default:
      print("Not implemented packet type: ${type}");
      break;
  }
});
```

Packet structure would be like:
* 2 byte integer as packet length,
* 2 byte integer as packet type,
* length times bytes for packet data,
and ensured one:
* 1 byte magic value header (to allow skip to it if failed),
* 2 byte integer as packet length,
* 2 byte integer as packet length checksum (simple negation would be fine),
* 2 byte integer as packet type,
* length times bytes for packet data,
* 2 byte integer as checksum of type and data,

PsychoXIVI, 2019-07-20:
I had a moment to think about implementing this, and there is next version of my proposal.
Changes:
+ use packet objects that implements some packet interface; then make all of these registerable to receive for this connection.
+ `receive`/`send` and `receiveRaw`/`sendRaw`. First pair for packets objects, second pair for raw packet ID/bytes.
+ move whole packets utilities to other import file, since not every application would like to use this a bit more complicated thing.

Note that `receiveRaw`/`sendRaw` behaves like in my previous comment above (switching action by ID to handle and sending right ID and bytes - both to be done by user).

User code should look like:
```dart
import 'package:flutter_bluetooth_serial/packets.dart';

class SomeDataPacket extends Packet
{
  final packetID = 0x01;
  final Uint8List bytes;

  /// Used internally to load bytes to retrive object later from.
  SomeDataPacket(Uint8List bytes) : this.bytes = bytes;

  /// Prepares packet for sending `SomeData`.
  SomeDataPacket.from(SomeData object) :
    bytes = (/* do encode object into bytes */null);

  /// Retrives `SomeData` from packet.
  SomeData get() {
    return SomeData(/* decode bytes into object */);
  }
}

//...

BluetoothConnectionAsPackets packets = BluetoothConnectionAsPackets.fromConnection(connection);

packets.receive<EchoRequestPacket>((packet) {
  packets.send(EchoResultPacket.forResponseTo(packet));
});
packets.receive<SomeDataPacket>((packet) {
  someDataStream.add(packet.get());
  // ^ user could handle packet here or delegate it (i.e. trough nice stream)
});
packets.receive<RequestSomeDataPacket>((packet) {
  packets.send(SomeDataPacket.from(someDataObject));
});
packets.receiveDefault((Packet packet) {
  print("Not implemented packet type: ${packet.packetID}");
});
```

Big thanks for `Miyoyo#5957` and `ping#0001` from Flutter Discord community.

PsychoXIVI, 2019-07-21:
After another while, I see that such delegating serialization and deserialization seems to be unnecessarily boiler-plate for utility like that. I user wants to use it, there will be no problem with previous design, but having whole registering, multiple types and so on its quite too have restriction for simple utility.

I think finally it could stay like this:
```
import 'package:flutter_bluetooth_serial/packets.dart';

/// User defined enum-like class for packets IDs.
class PacketType {
  static const Acknowledgement    = 0x00;
  static const SomeOlderAPIPacket = 0x01;
  static const SampleDataRequest  = 0x02;
  static const SampleData         = 0x03;
  static const SomeDataRequest    = 0x16;
  static const SomeData           = 0x17;
  static const EchoRequestPacket  = 0xE0;
  static const EchoResponsePacket = 0xE1;
  //...
}

//...

BluetoothConnectionAsPackets packets = BluetoothConnectionAsPackets.fromConnection(connection);

packets.receive((int type, Uint8List data) {
  switch (type) {
    case PacketType.Acknowledgement:
      // Ignore;
      break;
    case PacketType.SampleDataRequest:
      dataSamplesStream.add(DataSample.fromBytes(data));
      // ^ user could handle packet here or delegate it (i.e. trough nice stream)
      break;
    case PacketType.SomeOlderAPIPacket:
      // Deprecated types IDs should not be reused in real environment
      break;
    case PacketType.EchoRequestPacket:
      packets.send(PacketType.EchoResponsePacket, data); // or `connection.asPackets.send(0x04, data);`
      break;
    case PacketType.SomeDataRequest:
      packets.send(PacketType.SomeData, someDataObject.toBytes())
      break;
    //...
    default:
      print("Not implemented packet type: ${type}");
      break;
  }
});
```

PsychoXIVI, 2019-09-11:
Progress: https://github.com/PsychoXIVI/flutter_bluetooth_serial/commit/bcefb62f7a7cb57ff8b642d371388434dbcdec77

Erhannis, 2023-06-15:
This conversation reminds me of the MessageSocket class I wrote.
https://github.com/Erhannis/flutter_zeroconnect/blob/master/lib/MessageSocket.dart
It wraps a Socket, you hand it Strings or byte arrays, and it prefixes them with a length (and simple length checksum) and sends the bytes.  You request a string/bytes, and it reads/verifies a length, then that many bytes, and returns them.  The checksum isn't really sufficient for lossy channels, but it helps avoid accidentally waiting for millions of bytes.  (See also `maxLen`.)  There's also a 1-way ping function, just a plain length int of -1.
(The readme shows usage.  https://github.com/Erhannis/flutter_zeroconnect )