edufolly/flutter_bluetooth_serial #45: Connect Fails with exception S7 Edge

sajal-sh, 2019-07-08:
In the example code, after device search: OnTap function
I tried calling below:
onTap: () {
                print(">>>OnTap");
                
                //try to connect
                final connection =  BluetoothConnection.toAddress(result.device.address);
                connection.then((value)=> print('Connected to the device'))
                          .catchError((error)=> print("Error: "+error));
                
                
                //Navigator.of(context).pop(result.device);

Commented the navigator pop

Exception has occurred.
PlatformException (PlatformException(connect_error, read failed, socket might closed or timeout, read ret: -1, java.io.IOException: read failed, socket might closed or timeout, read ret: -1
	at android.bluetooth.BluetoothSocket.readAll(BluetoothSocket.java:821)
	at android.bluetooth.BluetoothSocket.readInt(BluetoothSocket.java:833)
	at android.bluetooth.BluetoothSocket.connect(BluetoothSocket.java:458)
	at io.github.edufolly.flutterbluetoothserial.BluetoothConnection.connect(BluetoothConnection.java:57)
	at io.github.edufolly.flutterbluetoothserial.BluetoothConnection.connect(BluetoothConnection.java:64)
	at io.github.edufolly.flutterbluetoothserial.FlutterBluetoothSerialPlugin.lambda$onMethodCall$0(FlutterBluetoothSerialPlugin.java:880)
	at io.github.edufolly.flutterbluetoothserial.-$$Lambda$FlutterBluetoothSerialPlugin$ysyByl051NlOhDx-4TeZKtqb1pA.run(Unknown Source:10)
	at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:245)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1162)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:636)
	at java.lang.Thread.run(Thread.java:764)
))

sajal-sh, 2019-07-08:
//DiscoveryPage.dart

 @override
  Widget build(BuildContext context) {
    return Scaffold(
        appBar: AppBar(
          title: isDiscovering
              ? Text('Discovering devices')
              : Text('Discovered devices'),
          actions: <Widget>[
            (isDiscovering
                ? FittedBox(
                    child: Container(
                        margin: new EdgeInsets.all(16.0),
                        child: CircularProgressIndicator(
                            valueColor:
                                AlwaysStoppedAnimation<Color>(Colors.white))))
                : IconButton(
                    icon: Icon(Icons.replay), onPressed: _restartDiscovery))
          ],
        ),
        body: ListView.builder(
          itemCount: results.length,
          itemBuilder: (BuildContext context, index) {
            BluetoothDiscoveryResult result = results[index];
            return BluetoothDeviceListEntry(
              device: result.device,
              rssi: result.rssi,
              onTap: () {
                print(">>>OnTap");
                
                //try to connect
                final connection =  BluetoothConnection.toAddress(result.device.address);
                connection.then((value)=> print('Connected to the device'))
                          .catchError((error)=> print("Error: "+error));
                
                
                //Navigator.of(context).pop(result.device);
              },
            );
          },
        ));
  }
}

PsychoXIVI, 2019-07-09:
Duplicate of #18

closed 2019-07-09 by PsychoXIVI