edufolly/flutter_bluetooth_serial #46: dispose problem

JAICHANGPARK, 2019-07-12:
Hi 
I have some problem in dispose process;

I tried to disconnection process using button 

`
`

```
Tried to send a platform message to Flutter, but FlutterJNI was detached from native C++. Could not send. Channel: flutter_bluetooth_serial/read/1. Response ID: 0
```

PsychoXIVI, 2019-07-12:
You used right in your case function, `bluetoothConnection.finish` which allows to close the connection gracefully - the remaining bytes of output stream should be sent to remote device. 

_There is also `bluetoothConnection.cancel` which does similar thing, but there is not guarantee that all remaining bytes of output stream would be sent. _

```
Tried to send a platform message to Flutter, but FlutterJNI was detached from native C++. Could not send. Channel: flutter_bluetooth_serial/read/1. Response ID: 0
```
This message might be thrown because of incoming data to your Flutter application from the native connection implementation that is in middle of being closed (exactly speaking, between disconnecting native socket and detaching incoming data stream to Flutter (and freeing the connection data)).

Is this message logged every time you run `.finish`?

PsychoXIVI, 2019-08-03:
@JAICHANGPARK , closing due to lack of response.

If you found solution - please share it for further readers.

Feel free to comment - the issue could be reopened if necessary.

closed 2019-08-03 by PsychoXIVI