edufolly/flutter_bluetooth_serial #47: Build issue

NourEldinShobier, 2019-07-22:
I keep getting this error when i try to build the application, any idea why?

```
C:\flutter\.pub-cache\hosted\pub.dartlang.org\flutter_bluetooth_serial-0.2.0\android\src\main\java\io\github\edufolly\flutterbluetoothserial\FlutterBluetoothSerialPlugin.java:13: error: package androidx.core.app does not exist
import androidx.core.app.ActivityCompat;
                        ^
C:\flutter\.pub-cache\hosted\pub.dartlang.org\flutter_bluetooth_serial-0.2.0\android\src\main\java\io\github\edufolly\flutterbluetoothserial\FlutterBluetoothSerialPlugin.java:14: error: package androidx.core.content does not exist
import androidx.core.content.ContextCompat;
                            ^
C:\flutter\.pub-cache\hosted\pub.dartlang.org\flutter_bluetooth_serial-0.2.0\android\src\main\java\io\github\edufolly\flutterbluetoothserial\FlutterBluetoothSerialPlugin.java:203: error: cannot find symbol
                                                ActivityCompat.startActivity(registrar.activity(), intent, null);
```

closed 2019-07-22 by NourEldinShobier

PsychoXIVI, 2019-07-22:
You can post your solution after closing - in some more people could find it if necessary.

NourEldinShobier, 2019-07-23:
I need to add these 2 line in `gradle.properties`
```
android.enableJetifier=true
android.useAndroidX=true
```