edufolly/flutter_bluetooth_serial #49: discovery time can set?

scofieldpeng, 2019-07-27:
Hi, can add a feature can user can customize the discovery timeout? the default timeout time is too short that can not find all ble devices

PsychoXIVI, 2019-07-27:
There is no customizable discovery time built-in. This is very simple thing, I am not sure it should go into library, but I will think about it.

For now can make use of future/promise race (paralel timeout and discovering which both cancel each other), it should be like few lines of code. 

_I will write it in few hours and post here for people who are looking for answer later._

scofieldpeng, 2019-08-06:
> There is no customizable discovery time built-in. ~This is very simple thing, I am not sure it should go into library, but I will think about it.~ EMM, there is a problem: I don't think we can extend discovery time :C Android API seems to miss that functionality.
> 
> From Android docs:
> 
> > To start discovering devices, simply call `startDiscovery()`. The process is asynchronous and returns a boolean value indicating whether discovery has successfully started. The discovery process usually involves an inquiry scan of about 12 seconds, followed by a page scan of each device found to retrieve its Bluetooth name.
> 
> For limiting discovery time you can make use of future/promise race (paralel timeout and discovering which both cancel each other), it should be like few lines of code, but I cannot find something to extend this time...
> 
> Maybe running multiple times every few seconds `startDiscovery` might help? I will test that in few days on my own, but if you get the answer, please notify me here.
> 
> BTW: Note that Bluetooth Low Energy devices might be undiscoverable on some devices.

thanks for reply, i search the document and slove this by search multiple times, thanks a million:-)

edufolly, 2020-12-04:
If you have further questions, please reopen this issue or open a new one.

Best.

closed 2020-12-04 by edufolly