edufolly/flutter_bluetooth_serial #50: Connect after Disconnect Error

rickdroio, 2019-07-31:
Trying to make a simple test with this lib connecting to a BT printer.

```
BluetoothConnection connection = await BluetoothConnection.toAddress(printerAddress);
connection.output.add(utf8.encode(text)); //actually prints the text on printer
...
connection.cancel();
```

on log I got the message:

`D/FlutterBluePlugin(15353): Disconnected (id: 14)`

But when I try to execute the same code it gives me the follow error:

```
I/flutter (15353): PlatformException(connect_error, read failed, socket might closed or timeout, read ret: -1, java.io.IOException: read failed, socket might closed or timeout, read ret: -1
I/flutter (15353): 	at android.bluetooth.BluetoothSocket.readAll(BluetoothSocket.java:774)
I/flutter (15353): 	at android.bluetooth.BluetoothSocket.readInt(BluetoothSocket.java:788)
I/flutter (15353): 	at android.bluetooth.BluetoothSocket.connect(BluetoothSocket.java:411)
I/flutter (15353): 	at io.github.edufolly.flutterbluetoothserial.BluetoothConnection.connect(BluetoothConnection.java:57)
I/flutter (15353): 	at io.github.edufolly.flutterbluetoothserial.BluetoothConnection.connect(BluetoothConnection.java:64)
I/flutter (15353): 	at io.github.edufolly.flutterbluetoothserial.FlutterBluetoothSerialPlugin.lambda$onMethodCall$0(FlutterBluetoothSerialPlugin.java:880)
I/flutter (15353): 	at io.github.edufolly.flutterbluetoothserial.-$$Lambda$FlutterBluetoothSerialPlugin$ysyByl051NlOhDx-4TeZKtqb1pA.run(Unknown Source:10)
I/flutter (15353): 	at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:245)
I/flutter (15353): 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
```

Looks like it doesnt close the connection.
Any clue how to fix it?

Thanks

PsychoXIVI, 2019-07-31:
The error is about connection, rather than canceling or disconnecting. See #18 and especially [my last comment there](https://github.com/edufolly/flutter_bluetooth_serial/issues/18#issuecomment-506472861), where I listed possible problems causing the issue.

If you are sure that all the problems listed in the comment are not your case, please feel free to ask for further help here.

BTW, 
> when I try to execute the same code
Does it mean, that you are immediately creating another connection just after closing the previous one? You might want to `await connection.cancel();` in that case, or maybe even add some more delay.

rickdroio, 2019-08-02:
Dear @PsychoXIVI 
This plugin works as expected with my BT printer during the entire object lifecycle except
close connection and try to create a new connection in another page.
FYI when I turn off the BT the plugin returns "Disconnected from remote" even after object disposal.
is there any guideline for this situation?

is there

rickdroio, 2019-08-02:
> when I try to execute the same code
> Does it mean, that you are immediately creating another connection just after closing the previous one? You might want to await connection.cancel(); in that case, or maybe even add some more delay.

actually it never get out of this await connection.cancel();

PsychoXIVI, 2019-08-03:
> actually it never get out of this await connection.cancel();

I will look into that closely in few hours.

rickdroio, 2019-08-05:
Dear @PsychoXIVI 

Using the official repository I got the follow response:

```
I/flutter (15895): Disconnected by remote request
I/flutter (15895): TEST: connection cancelled
D/FlutterBluePlugin(15895): Disconnected (id: 3)
```

But actually it doesnt disconnect it. When I try to reconnect after few minutes I got the message:

```
I/flutter (15895): Cannot connect, exception occured
I/flutter (15895): PlatformException(connect_error, read failed, socket might closed or timeout, read ret: -1, java.io.IOException: read failed, socket might closed or timeout, read ret: -1
I/flutter (15895): 	at android.bluetooth.BluetoothSocket.readAll(BluetoothSocket.java:774)
I/flutter (15895): 	at android.bluetooth.BluetoothSocket.readInt(BluetoothSocket.java:788)
I/flutter (15895): 	at android.bluetooth.BluetoothSocket.connect(BluetoothSocket.java:411)
I/flutter (15895): 	at io.github.edufolly.flutterbluetoothserial.BluetoothConnection.connect(BluetoothConnection.java:57)
I/flutter (15895): 	at io.github.edufolly.flutterbluetoothserial.BluetoothConnection.connect(BluetoothConnection.java:64)
I/flutter (15895): 	at io.github.edufolly.flutterbluetoothserial.FlutterBluetoothSerialPlugin.lambda$onMethodCall$0(FlutterBluetoothSerialPlugin.java:880)
```

Using your fixed branch it fix the issue!!! Great work!!!
Thanks for your support!!

Do you have any plan to release new plugin version?

Thank you again!!

PsychoXIVI, 2019-08-05:
That sound great. I think I understand what went wrong: There were #48 PR fixing some disconnection problem - and it isn't already at official package release. 

I will prepare v0.2.1 ASAP, to include that important fix. I'm sorry for procrastination.

PsychoXIVI, 2019-08-05:
Published patch version 0.2.1. You should update your `pubspec.yaml`. 

I am closing the issue by now. Let me know in case is there something wrong with the patch version.

closed 2019-08-05 by PsychoXIVI