edufolly/flutter_bluetooth_serial #52: getBondedDevices throws [Failed resolution of: Landroidx/core/content/ContextCompat];

n00ne1mportant, 2019-08-08:
Hello!
The `getBondedDevices` call never finishes, it crashes the app.
I'm trying to get now connected device's address. I've gone through the lib code and failed to find this method anywhere so my idea was to get all the paired devices and find the one with BluetoothBondState.bound. IDK if this method gets paired devices or connected devices, this is what I'm testing / trying to figure out.
However... the initial call fails. This is me testing the method:
`Future isConnected() async {
    var devices = await FlutterBluetoothSerial.instance.getBondedDevices();
    print(devices.length);
    return devices.length > 1;
  }`
I know it makes no sense atm, I just want it to return without crashing for now.
`class _MyAppState extends State<MyApp> {
  @override
  Widget build(BuildContext context) {
    BluetoothController.instance().isConnected().then((r) => print(r));
    return MaterialApp(
      title: 'Test',
      home: Scaffold(
        body: Center(
        etc.......'`
Again, I know. Async in build sux. I just wanna see the output.
This is the debug console output doe:
`E/flutter ( 8028): [ERROR:flutter/shell/platform/android/platform_view_android_jni.cc(40)] java.lang.NoClassDefFoundError: Failed resolution of: Landroidx/core/content/ContextCompat;
E/flutter ( 8028): 	at io.github.edufolly.flutterbluetoothserial.FlutterBluetoothSerialPlugin.ensurePermissions(FlutterBluetoothSerialPlugin.java:988)
E/flutter ( 8028): 	at io.github.edufolly.flutterbluetoothserial.FlutterBluetoothSerialPlugin.onMethodCall(FlutterBluetoothSerialPlugin.java:760)
E/flutter ( 8028): 	at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler.onMessage(MethodChannel.java:222)
E/flutter ( 8028): 	at io.flutter.embedding.engine.dart.DartMessenger.handleMessageFromDart(DartMessenger.java:96)
E/flutter ( 8028): 	at io.flutter.embedding.engine.FlutterJNI.handlePlatformMessage(FlutterJNI.java:643)
E/flutter ( 8028): 	at android.os.MessageQueue.nativePollOnce(Native Method)
E/flutter ( 8028): 	at android.os.MessageQueue.next(MessageQueue.java:326)
E/flutter ( 8028): 	at android.os.Looper.loop(Looper.java:181)
E/flutter ( 8028): 	at android.app.ActivityThread.main(ActivityThread.java:7156)
E/flutter ( 8028): 	at java.lang.reflect.Method.invoke(Native Method)
E/flutter ( 8028): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:494)
E/flutter ( 8028): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:975)
E/flutter ( 8028): Caused by: java.lang.ClassNotFoundException: Didn't find class "androidx.core.content.ContextCompat" on path: DexPathList[[zip file "/data/app/com.example.flutter_controler-38NJRdIAA15Ftz9Zg0Y2IA==/base.apk"],nativeLibraryDirectories=[/data/app/com.example.flutter_controler-38NJRdIAA15Ftz9Zg0Y2IA==/lib/arm, /data/app/com.example.flutter_controler-38NJRdIAA15Ftz9Zg0Y2IA==/base.apk!/lib/armeabi-v7a, /system/lib, /system/vendor/lib]]
E/flutter ( 8028): 	at dalvik.system.BaseDexClassLoader.findClass(BaseDexClassLoader.java:134)
E/flutter ( 8028): 	at java.lang.ClassLoader.loadClass(ClassLoader.java:379)
E/flutter ( 8028): 	at java.lang.ClassLoader.loadClass(ClassLoader.java:312)
E/flutter ( 8028): 	... 12 more
E/flutter ( 8028):
F/flutter ( 8028): [FATAL:flutter/shell/platform/android/platform_view_android_jni.cc(77)] Check failed: CheckException(env).
F/libc    ( 8028): Fatal signal 6 (SIGABRT), code -6 (SI_TKILL) in tid 8028 (utter_controler), pid 8028 (utter_controler)
*** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
Build fingerprint: 'samsung/j7y17ltexx/j7y17lte:9/PPR1.180610.011/J730FXXU4CSF1:user/release-keys'
Revision: '7'
ABI: 'arm'
pid: 8028, tid: 8028, name: utter_controler  >>> com.example.flutter_controler <<<
signal 6 (SIGABRT), code -6 (SI_TKILL), fault addr --------
Abort message: '[FATAL:flutter/shell/platform/android/platform_view_android_jni.cc(77)] Check failed: CheckException(env).`

The manifest:
`<manifest xmlns:android="http://schemas.android.com/apk/res/android"
 xmlns:tools="http://schemas.android.com/tools"
 package="com.example.flutter_controler">
    
    <uses-sdk tools:overrideLibrary="io.github.edufolly.flutterbluetoothserial"/>
    <uses-permission android:name="android.permission.INTERNET"/>
    <uses-permission android:name="android.permission.BLUETOOTH" />
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />

</manifest>`
Tried `flutter clean`. Breakpoint just at the call says it has to be that call that breaks the app. Commenting it out works...
2 hours of googlng didn't help.
I'm running linux (obvious from debug message), live device Samsung J7, Android 9.1.
Any ideas?

PsychoXIVI, 2019-08-08:
Hello,

It looks like the main issue is unavailable resolution of AndroidX. 

Which version of the plugin are you using? If still `0.2`, please update to `0.2.1` - there was a some fix related to AndroidX afik.

n00ne1mportant, 2019-08-09:
Thank you for quick response.
I am using ´0.2.1´. Not using AndroidX. I opened the project with Android Studio to move it to AndroidX via the refactor menu. It gave me "Not using" message and the directory was unchanged.
Is it possible I'm using it the wrong way?
I just don't see it atm might be smtg trivial.
Ideas?
The package file is bellow:
---------
version: 1.0.0+1

environment:
  sdk: ">=2.1.0 <3.0.0"

dependencies:
  flutter:
    sdk: flutter

  # The following adds the Cupertino Icons font to your application.
  # Use with the CupertinoIcons class for iOS style icons.
  cupertino_icons: ^0.1.2
  flutter_bluetooth_serial: ^0.2.1

dev_dependencies:
  flutter_test:
    sdk: flutter

n00ne1mportant, 2019-08-09:
Adding 
```
android.useAndroidX=true
android.enableJetifier=true
```
to gradle.properties fixes it :man_facepalming: 

Maybe we could add a note to Readme? It was a solution to similar compile time issue as well. 
Anyways I'll close it now

closed 2019-08-09 by n00ne1mportant

PsychoXIVI, 2019-08-09:
I was just about to ask about it :D haha

kamilk91, 2019-10-15:
@PsychoXIVI the same problem with v0.2.2, and adding 
```
android.useAndroidX=true
android.enableJetifier=true
```
helped.