edufolly/flutter_bluetooth_serial #54: requestEnable crashes app

NicolasSimmonds, 2019-08-12:
Hello, trying out this package and wanted to turn on bluetooth with a button press, but when I do it the app just crashes every time. 
Here is the code for that

```
RaisedButton(
                onPressed: () {
                  future() async {
                    await FlutterBluetoothSerial.instance.requestEnable();
                  }

                  future().then((_) => print('enabled'));
                },
              ),
```

NicolasSimmonds, 2019-08-12:
OK, I fixed the issue but I dont understand why:

In the gradle.properties file (projectname>android>gradle.properties) I addes these two lines
`android.enableJetifier=true
android.useAndroidX=true`

and after that the the app works as expected, but I dont understand why, any idea?

PsychoXIVI, 2019-08-12:
The plugin uses AndroidX for sometime. Newest version might enforce usage of these also for user app. I will experiment more with this, since it's like second similar issue 🤔

closed 2019-08-19 by PsychoXIVI

helloitsm3, 2019-11-23:
Hi, I have the exact same gradle version as your example but I can't seem to get this to work. It keeps crashing my app but when I try running your app, it doesn't seem to have any issue. Do you have any fix for this when trying to turn on bluetooth? @PsychoXIVI