edufolly/flutter_bluetooth_serial #55: when discovery,error occur in somecases like Key android.bluetooth.device.extra.RSSI expected Short but value was a java.lang.Integer.

scofieldpeng, 2019-08-12:
Hi, sometimes when i discovery the devices, the console will print these error, it seems that the type assert error, below is error snapshot:

```
W/Bundle  ( 8928): Key android.bluetooth.device.extra.RSSI expected Short but value was a java.lang.Integer.  The default value -32768 was returned.
W/Bundle  ( 8928): Attempt to cast generated internal exception:
W/Bundle  ( 8928): java.lang.ClassCastException: java.lang.Integer cannot be cast to java.lang.Short
W/Bundle  ( 8928): 	at android.os.BaseBundle.getShort(BaseBundle.java:839)
W/Bundle  ( 8928): 	at android.os.Bundle.getShort(Bundle.java:733)
W/Bundle  ( 8928): 	at android.content.Intent.getShortExtra(Intent.java:6157)
W/Bundle  ( 8928): 	at io.github.edufolly.flutterbluetoothserial.FlutterBluetoothSerialPlugin$4.onReceive(FlutterBluetoothSerialPlugin.java:328)
W/Bundle  ( 8928): 	at android.app.LoadedApk$ReceiverDispatcher$Args.run(LoadedApk.java:1122)
W/Bundle  ( 8928): 	at android.os.Handler.handleCallback(Handler.java:755)
W/Bundle  ( 8928): 	at android.os.Handler.dispatchMessage(Handler.java:95)
W/Bundle  ( 8928): 	at android.os.Looper.loop(Looper.java:154)
W/Bundle  ( 8928): 	at android.app.ActivityThread.main(ActivityThread.java:6121)
W/Bundle  ( 8928): 	at java.lang.reflect.Method.invoke(Native Method)
W/Bundle  ( 8928): 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:905)
W/Bundle  ( 8928): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:795)
```

PsychoXIVI, 2019-08-12:
Which version of Android are you using?

scofieldpeng, 2019-08-14:
sorry for reply too late. i use the tablet which is arm (32 bit) and Android 7.1.2, i'm not sure is the arm platform problem or not.

PsychoXIVI, 2019-08-14:
> sometimes when i discovery the devices

Is there device, which while detected causes the problem? Or is it for all discovered devices?

Anyway, I will try prepare some branch with more debugging for this issue tonight.

scofieldpeng, 2019-08-15:
> > sometimes when i discovery the devices
> 
> Is there device, which while detected causes the problem? Or is it for all discovered devices?
> 
> Anyway, I will try prepare some branch with more debugging for this issue tonight.

not all discovered devices but some of them

PsychoXIVI, 2019-08-15:
I prepared some more debugging code for this issue:
https://github.com/PsychoXIVI/flutter_bluetooth_serial/commit/d9d72ceb6586720481a89a7273039e05be849017

My system seems to be fine with passing short value for RSSI extra on `ACTION_FOUND`. For me discovery related log looks like this:
```
D/FlutterBluePlugin(22373): Starting discovery
D/FlutterBluePlugin(22373): Discovered XX:XX:XX:XX:XX:XX (PSYCHOX-EARTH)
D/FlutterBluePlugin(22373): Extra RSSI field is short!
D/FlutterBluePlugin(22373): Valid RSSI of value -44, passing to common code
D/FlutterBluePlugin(22373): Discovered XX:XX:XX:XX:XX:XX
```

This code will try to use both short and integer for RSSI value. It might get rid of the whole warning, but **please post your log** using this fix, so I can confirm that it is working correctly.

You can use it temporary using this dependency declaration for the plugin in your app:
```yaml
dependencies:
  flutter_bluetooth_serial:
    git:
      url: git://github.com/PsychoXIVI/flutter_bluetooth_serial.git
      ref: rssi_issue_55
```

It will be merged via pull request to main repository if the fix is working. I will mention you in such case, to remind you change back te dependency declaration back to main repo.

scofieldpeng, 2019-08-16:
OK, i will try it and check the result

scofieldpeng, 2019-08-16:
i test it in my app, no error occur,tks a million。but i found another problem about discovery. when i call the discovery method, the terminal output these log:

```
D/FlutterBluePlugin(10986): Canceling discovery (stream closed)
I/flutter (10986): scan ble printer start
D/FlutterBluePlugin(10986): Canceling discovery
D/FlutterBluePlugin(10986): Starting discovery
```

and it seems like not discovery, then i call the cancelDiscovery method and continue to call the discovery method, the same log occur.

finally, I find if I call the disable and then call the enable method, now when i call the discovery method, everything is ok.

i have checked my codes and everything is follow the you doc and don't know how to fix this problem. This case is always happen after I run the `flutter run` for sometimes repeatedly. Can u give me some advice about it? tks a million:-)

PsychoXIVI, 2019-08-16:
I would like to see input for whole discovering process, as I mentioned earlier. It should contains something like "`Extra RSSI field is integer!`" in your case as far I understand.

> the terminal output these log:
> ```
> D/FlutterBluePlugin(10986): Canceling discovery (stream closed)
> I/flutter (10986): scan ble printer start
> D/FlutterBluePlugin(10986): Canceling discovery
> D/FlutterBluePlugin(10986): Starting discovery
> ```

Please note that every start of discovery might first try close current ongoing discovery, that's why calling `startDiscovery` makes it first `Canceling discovery`.

Whole discovery process log for me looks like this:
```
Starting discovery
D/FlutterBluePlugin(22373): Discovered XX:XX:XX:XX:XX:XX (NAME-OF-DEVICE-FOUND-HERE)
D/FlutterBluePlugin(22373): Extra RSSI field is short!
D/FlutterBluePlugin(22373): Valid RSSI of value -44, passing to common code
D/FlutterBluePlugin(22373): Discovered XX:XX:XX:XX:XX:XX
D/FlutterBluePlugin(22373): Discovery finished
D/FlutterBluePlugin(22373): Canceling discovery (stream closed)
```

Do you have `Discovery finished` in your log after starting the discovery?

scofieldpeng, 2019-08-17:
yes, in normal case, it will output these logs, but sometimes(i only face 3 times these days) it won't output anything after show `Start discovery`. After i close and reopen the bluetooth, everything is ok.

PsychoXIVI, 2019-08-17:
I would like to see input for whole discovering process, as I mentioned earlier twice. Please, post it here. You can even post your whole long log if you are not sure what I am talking about (but use [Pastebin](https://pastebin.com) or something similar then).

PsychoXIVI, 2019-10-14:
Closing due to no response. Please feel free to comment, I would reopen the issue in such case.

closed 2019-10-14 by PsychoXIVI