edufolly/flutter_bluetooth_serial #58: Example App crashes

hendri-tobing, 2019-08-17:
I have install the example app on Xiaomi Redmi note 4. If I click any buttons, the app crashes. 

🔥  To hot reload changes while running, press "r". To hot restart (and rebuild state), press "R".
An Observatory debugger and profiler on Redmi Note 4 is available at: http://127.0.0.1:51495/SnxI_e9IxJI=/
For a more detailed help message, press "h". To detach, press "d"; to quit, press "q".
V/BoostFramework(11199): BoostFramework() : mPerf = com.qualcomm.qti.Performance@689460a
E/flutter (11199): [ERROR:flutter/shell/platform/android/platform_view_android_jni.cc(40)] java.lang.NoClassDefFoundError: Failed resolution of: Landroidx/core/app/ActivityCompat;
E/flutter (11199):      at io.github.edufolly.flutterbluetoothserial.FlutterBluetoothSerialPlugin.onMethodCall(FlutterBluetoothSerialPlugin.java:433)
E/flutter (11199):      at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler.onMessage(MethodChannel.java:222)
E/flutter (11199):      at io.flutter.embedding.engine.dart.DartMessenger.handleMessageFromDart(DartMessenger.java:96)
E/flutter (11199):      at io.flutter.embedding.engine.FlutterJNI.handlePlatformMessage(FlutterJNI.java:643)
E/flutter (11199):      at android.os.MessageQueue.nativePollOnce(Native Method)
E/flutter (11199):      at android.os.MessageQueue.next(MessageQueue.java:329)
E/flutter (11199):      at android.os.Looper.loop(Looper.java:142)
E/flutter (11199):      at android.app.ActivityThread.main(ActivityThread.java:6375)
E/flutter (11199):      at java.lang.reflect.Method.invoke(Native Method)
E/flutter (11199):      at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:912)
E/flutter (11199):      at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:802)
E/flutter (11199): Caused by: java.lang.ClassNotFoundException: Didn't find class "androidx.core.app.ActivityCompat" on path: DexPathList[[zip file "/data/app/com.example.flutter_bluetooth_serial_example-1/base.apk"],nativeLibraryDirectories=[/data/app/com.example.flutter_bluetooth_serial_example-1/lib/arm64, /data/app/com.example.flutter_bluetooth_serial_example-1/base.apk!/lib/arm64-v8a, /system/lib64, /vendor/lib64]]
E/flutter (11199):      at dalvik.system.BaseDexClassLoader.findClass(BaseDexClassLoader.java:74)
E/flutter (11199):      at java.lang.ClassLoader.loadClass(ClassLoader.java:380)
E/flutter (11199):      at java.lang.ClassLoader.loadClass(ClassLoader.java:312)
E/flutter (11199):      ... 11 more
E/flutter (11199):
F/flutter (11199): [FATAL:flutter/shell/platform/android/platform_view_android_jni.cc(77)] Check failed: CheckException(env).
F/libc    (11199): Fatal signal 6 (SIGABRT), code -6 in tid 11199 (_serial_example)
*** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
Build fingerprint: 'xiaomi/mido/mido:7.0/NRD90M/V10.2.3.0.NCFMIXM:user/release-keys'
Revision: '0'
ABI: 'arm64'
pid: 11199, tid: 11199, name: _serial_example  >>> com.example.flutter_bluetooth_serial_example <<<
signal 6 (SIGABRT), code -6 (SI_TKILL), fault addr --------
Abort message: '[FATAL:flutter/shell/platform/android/platform_view_android_jni.cc(77)] Check failed: CheckException(env).
'
    x0   0000000000000000  x1   0000000000002bbf  x2   0000000000000006  x3   0000000000000008
    x4   0000000000000036  x5   0000000000800000  x6   0000007f82cfa000  x7   0000000000000000
    x8   0000000000000083  x9   ffffffffffffffdf  x10  0000000000000000  x11  0000000000000001
    x12  ffffffffffffffff  x13  0000000000000000  x14  0000000000000000  x15  000a3ad29ece668e
    x16  0000007f81d45ed0  x17  0000007f81cef510  x18  0000000072090ddc  x19  0000007f82daeb48
    x20  0000000000000006  x21  0000007f82daeaa0  x22  000000000000000b  x23  0000007f55e4d1d8
    x24  0000000000000031  x25  0000000000000001  x26  0000007f71738520  x27  0000007f7172b1a8
    x28  0000007f7f6f8980  x29  0000007fd65fdc60  x30  0000007f81cec9a0
    sp   0000007fd65fdc40  pc   0000007f81cef518  pstate 0000000060000000
backtrace:
    #00 pc 000000000006b518  /system/lib64/libc.so (tgkill+8)
    #01 pc 000000000006899c  /system/lib64/libc.so (pthread_kill+64)
    #02 pc 0000000000023ee8  /system/lib64/libc.so (raise+24)
    #03 pc 000000000001c96c  /system/lib64/libc.so (abort+52)
    #04 pc 0000000000d34ca8  /data/app/com.example.flutter_bluetooth_serial_example-1/lib/arm64/libflutter.so (offset 0xd20000)
    #05 pc 0000000000d2778c  /data/app/com.example.flutter_bluetooth_serial_example-1/lib/arm64/libflutter.so (offset 0xd20000)
    #06 pc 0000000000d25cb0  /data/app/com.example.flutter_bluetooth_serial_example-1/lib/arm64/libflutter.so (offset 0xd20000)
    #07 pc 0000000000d6c294  /data/app/com.example.flutter_bluetooth_serial_example-1/lib/arm64/libflutter.so (offset 0xd20000)
    #08 pc 0000000000d35754  /data/app/com.example.flutter_bluetooth_serial_example-1/lib/arm64/libflutter.so (offset 0xd20000)
    #09 pc 0000000000d3a45c  /data/app/com.example.flutter_bluetooth_serial_example-1/lib/arm64/libflutter.so (offset 0xd20000)
    #10 pc 000000000001852c  /system/lib64/libutils.so (_ZN7android6Looper9pollInnerEi+1032)
    #11 pc 0000000000018074  /system/lib64/libutils.so (_ZN7android6Looper8pollOnceEiPiS1_PPv+60)
    #12 pc 00000000000f76a8  /system/lib64/libandroid_runtime.so (_ZN7android18NativeMessageQueue8pollOnceEP7_JNIEnvP8_jobjecti+48)
    #13 pc 00000000022112b0  /system/framework/arm64/boot-framework.oat (offset 0x195e000) (android.os.MessageQueue.nativePollOnce+140)
    #14 pc 0000000002212f30  /system/framework/arm64/boot-framework.oat (offset 0x195e000) (android.os.MessageQueue.next+236)
    #15 pc 000000000220c8f8  /system/framework/arm64/boot-framework.oat (offset 0x195e000) (android.os.Looper.loop+340)
    #16 pc 0000000001a1d284  /system/framework/arm64/boot-framework.oat (offset 0x195e000) (android.app.ActivityThread.main+592)
    #17 pc 00000000000d2b68  /system/lib64/libart.so (art_quick_invoke_static_stub+600)
    #18 pc 00000000000df5b0  /system/lib64/libart.so (_ZN3art9ArtMethod6InvokeEPNS_6ThreadEPjjPNS_6JValueEPKc+252)
    #19 pc 000000000042991c  /system/lib64/libart.so (_ZN3artL18InvokeWithArgArrayERKNS_33ScopedObjectAccessAlreadyRunnableEPNS_9ArtMethodEPNS_8ArgArrayEPNS_6JValueEPKc+108)
    #20 pc 000000000042b48c  /system/lib64/libart.so (_ZN3art12InvokeMethodERKNS_33ScopedObjectAccessAlreadyRunnableEP8_jobjectS4_S4_m+1188)
    #21 pc 00000000003a3620  /system/lib64/libart.so (_ZN3artL13Method_invokeEP7_JNIEnvP8_jobjectS3_S3_+56)
    #22 pc 00000000006730f8  /system/framework/arm64/boot.oat (offset 0x5a2000) (java.lang.reflect.Method.invoke+180)
    #23 pc 00000000028e7440  /system/framework/arm64/boot-framework.oat (offset 0x195e000) (com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run+124)
    #24 pc 00000000028e8810  /system/framework/arm64/boot-framework.oat (offset 0x195e000) (com.android.internal.os.ZygoteInit.main+1532)
    #25 pc 00000000000d2b68  /system/lib64/libart.so (art_quick_invoke_static_stub+600)
    #26 pc 00000000000df5b0  /system/lib64/libart.so (_ZN3art9ArtMethod6InvokeEPNS_6ThreadEPjjPNS_6JValueEPKc+252)
    #27 pc 000000000042991c  /system/lib64/libart.so (_ZN3artL18InvokeWithArgArrayERKNS_33ScopedObjectAccessAlreadyRunnableEPNS_9ArtMethodEPNS_8ArgArrayEPNS_6JValueEPKc+108)
    #28 pc 0000000000429574  /system/lib64/libart.so (_ZN3art17InvokeWithVarArgsERKNS_33ScopedObjectAccessAlreadyRunnableEP8_jobjectP10_jmethodIDSt9__va_list+380)
    #29 pc 0000000000344acc  /system/lib64/libart.so (_ZN3art3JNI21CallStaticVoidMethodVEP7_JNIEnvP7_jclassP10_jmethodIDSt9__va_list+604)
    #30 pc 00000000000a782c  /system/lib64/libandroid_runtime.so
    #31 pc 00000000000a9f2c  /system/lib64/libandroid_runtime.so (_ZN7android14AndroidRuntime5startEPKcRKNS_6VectorINS_7String8EEEb+704)
    #32 pc 000000000000226c  /system/bin/app_process64
    #33 pc 000000000001a794  /system/lib64/libc.so (__libc_init+88)
    #34 pc 0000000000001ccc  /system/bin/app_process64

How to resolve this?

Thank you.

PsychoXIVI, 2019-08-17:
Are you using 0.2.1 version?

```
java.lang.NoClassDefFoundError: Failed resolution of: Landroidx/core/app/ActivityCompat;
```

The library uses AndroidX, so the app should too. Please try add 
```
android.useAndroidX=true
android.enableJetifier=true
```
to `gradle.properties` of your project.

hendri-tobing, 2019-08-18:
Yes, I am using 0.2.1 version. Just added those two lines to gradle.properties and it is working. Thank you.

closed 2019-08-18 by hendri-tobing