edufolly/flutter_bluetooth_serial #60: Problem disconnecting HC-05 module

bandido71, 2019-08-19:
Hello,

I've made a little app to test the HC-05 module, there is no problem when connecting, but when i tried to disconnect the module remains connected. I'm using the following code to disconnect, with a modification from #50 to test it,

```dart
           RaisedButton(
                padding: EdgeInsets.all(10.0),
                child: Text('Desconectar'),
                onPressed: () {
                  print('TEST: canceling the connection');
                  connection.cancel().then((_) {
                    print('TEST: connection cancelled');
                  });
                  Navigator.pop(context, true);
                },
              ), 
```
Only the first print statement is executed.

PsychoXIVI, 2019-08-19:
As you saw issue no. 50, I am assuming you are using 0.2.1 version.

I mentioned in the issue discussion adding more debug logs in implementation of `cancel` method. I prepared such modification: https://github.com/edufolly/flutter_bluetooth_serial/commit/8c376f80f4a51107a41bff3a8f8b93065cfae7d2

You can use it easily by changing your dependency to:
```
dependencies:
  flutter_bluetooth_serial:
    git:
      url: git://github.com/edufolly/flutter_bluetooth_serial.git
      ref: issue60_debug
```
_Note: that is temporary branch, after solving issue you should change it back to normal_

After applying the more logging, please share the log after using the `cancel` method.

bandido71, 2019-08-19:
I changed the dependency and i've got the following:
```
Restarted application in 2.420ms.
D/FlutterBluePlugin(10673): Connecting to 30:14:10:15:17:47 (id: 2)
W/BluetoothAdapter(10673): getBluetoothService() called with no BluetoothManagerCallback
I/flutter (10673): Connected to the device
I/flutter (10673): TEST: canceling the connection
I/flutter (10673): DEBUG: BluetoothConnection cancel - entered
I/flutter (10673): DEBUG: BluetoothConnection cancel - output closed
I/flutter (10673): DEBUG: BluetoothConnection cancel - entered
I/flutter (10673): DEBUG: BluetoothConnection cancel - output closed
```

Only output.close() executed twice, and the module still remains connected.

PsychoXIVI, 2019-08-19:
Are you listening to `connection.input` at the moment you are `cancel`ing the connection?

bandido71, 2019-08-19:
I'm not listening to conection.input, i was just testing connection an disconnection of the module.

PsychoXIVI, 2019-08-19:
Fix (as version 0.2.2) will land in hour I think.

PsychoXIVI, 2019-08-19:
@bandido71 Version 0.2.2 was published. Update your dependency declaration in `pubspec.yaml` and try it out. Please let me know whether it solves the issue for you.

bandido71, 2019-08-19:
With the new connection.close() the module disconnect without problems. Thanks you very much for your help.

closed 2019-08-19 by PsychoXIVI