edufolly/flutter_bluetooth_serial #61: "startDiscovery()" makes app crash Fatal signal 6 (SIGABRT),

earyzhe, 2019-08-19:
**When calling the "startDiscovery()" function to obtain discoverable devices, the app crashes with error:**

`
E/flutter (10018): [ERROR:flutter/shell/platform/android/platform_view_android_jni.cc(40)] java.lang.NoClassDefFoundError: Failed resolution of: Landroidx/core/content/ContextCompat;
E/flutter (10018): 	at io.github.edufolly.flutterbluetoothserial.FlutterBluetoothSerialPlugin.ensurePermissions(FlutterBluetoothSerialPlugin.java:988)
E/flutter (10018): 	at io.github.edufolly.flutterbluetoothserial.FlutterBluetoothSerialPlugin.onMethodCall(FlutterBluetoothSerialPlugin.java:789)
E/flutter (10018): 	at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler.onMessage(MethodChannel.java:222)
E/flutter (10018): 	at io.flutter.embedding.engine.dart.DartMessenger.handleMessageFromDart(DartMessenger.java:96)
E/flutter (10018): 	at io.flutter.embedding.engine.FlutterJNI.handlePlatformMessage(FlutterJNI.java:643)
E/flutter (10018): 	at android.os.MessageQueue.nativePollOnce(Native Method)
E/flutter (10018): 	at android.os.MessageQueue.next(MessageQueue.java:328)
E/flutter (10018): 	at android.os.Looper.loop(Looper.java:148)
E/flutter (10018): 	at android.app.ActivityThread.main(ActivityThread.java:6339)
E/flutter (10018): 	at java.lang.reflect.Method.invoke(Native Method)
E/flutter (10018): 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1084)
E/flutter (10018): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:945)
E/flutter (10018): Caused by: java.lang.ClassNotFoundException: Didn't find class "androidx.core.content.ContextCompat" on path: DexPathList[[zip file "/data/app/com.chargecontroller.android.abc-2/base.apk"],nativeLibraryDirectories=[/data/app/com.chargecontroller.android.abc-2/lib/arm, /data/app/com.chargecontroller.android.abc-2/base.apk!/lib/armeabi-v7a, /system/lib, /vendor/lib]]
E/flutter (10018): 	at dalvik.system.BaseDexClassLoader.findClass(BaseDexClassLoader.java:56)
E/flutter (10018): 	at java.lang.ClassLoader.loadClass(ClassLoader.java:380)
E/flutter (10018): 	at java.lang.ClassLoader.loadClass(ClassLoader.java:312)
E/flutter (10018): 	... 12 more
E/flutter (10018):
F/flutter (10018): [FATAL:flutter/shell/platform/android/platform_view_android_jni.cc(77)] Check failed: CheckException(env).
F/libc    (10018): Fatal signal 6 (SIGABRT), code -6 in tid 10018 (ler.android.abc)
Lost connection to device.
Exited (sigterm)
`

PsychoXIVI, 2019-08-19:
See https://github.com/edufolly/flutter_bluetooth_serial/issues/58#issuecomment-522228632. Please confirm if it does solve the issue.

earyzhe, 2019-08-19:
Yes that worked. Thanks

closed 2019-08-19 by PsychoXIVI