edufolly/flutter_bluetooth_serial #62: PlatformException when trying to connect to device.

earyzhe, 2019-08-19:
### Problem summary

### Steps to reproduce

When trying to connect to a device with an address a platform exception is thrown.

Steps to reproduce

1. Scan for devices to get a list.
2. Stop scanning for devices.
3. Try to connect to a device.

[image]
[image]

### Environment

```
Flutter 1.7.8+hotfix.4 • channel stable • https://github.com/flutter/flutter.git
Framework • revision 20e59316b8 (5 weeks ago) • 2019-07-18 20:04:33 -0700
Engine • revision fee001c93f
Tools • Dart 2.4.0
```

Using HC-10 BLE device.

This is working on IOS with Flutter Blue plugin.

PsychoXIVI, 2019-08-19:
See https://github.com/edufolly/flutter_bluetooth_serial/issues/18#issuecomment-506472861 . 

As far I understand, `HC-10` uses BLE, so your case is F):
> **F) You misunderstood Bluetooth Serial for Bluetooth Low Energy.**
> > These are quite [different protocols](https://www.google.com/search?q=bluetooth+ble+vs+serial), even while using same underlying band and link protocols (including almost the same pairing and discovery). This library is created for classic Serial connectivity - see [flutter_blue](https://pub.dev/packages/flutter_blue) for BLE instead.

BTW, classic serial connectivity might be unavailable on iOS, since it needs some certifications as far I remember.

_Closing for now, but feel free to comment more if necessary._

closed 2019-08-19 by PsychoXIVI

earyzhe, 2019-08-19:
Ah ok thanks.

SzymonGalazka, 2019-08-21:
I don't want to create a new thread so I ask here:
I got the same error while connecting to elm327 bluetooth device.
There are no explanations of the protocol used (see: https://yanosik.pl/data/support/instrukcja_connect.pdf) 
and I'm wondering if it's possible to use this package or flutter_blue to access this device.

vaibhiarora03, 2022-02-09:
hi @SzymonGalazka . were you able to resolve this issue?