edufolly/flutter_bluetooth_serial #63: cannot connect FATAL EXCEPTION

gillestasse, 2019-08-20:
When I try to connect using the sample on the README.MD, I get the following error:
```
E/AndroidRuntime(29972): java.lang.IllegalStateException: Reply already submitted
E/AndroidRuntime(29972): 	at io.flutter.embedding.engine.dart.DartMessenger$Reply.reply(DartMessenger.java:135)
E/AndroidRuntime(29972): 	at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler$1.error(MethodChannel.java:230)
E/AndroidRuntime(29972): 	at io.github.edufolly.flutterbluetoothserial.FlutterBluetoothSerialPlugin.lambda$connect$0(FlutterBluetoothSerialPlugin.java:260)
E/AndroidRuntime(29972): 	at io.github.edufolly.flutterbluetoothserial.-$$Lambda$FlutterBluetoothSerialPlugin$52C5RUgoP_r70sgAnP8ZXC1Ysz0.run(lambda)
E/AndroidRuntime(29972): 	at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:243)
E/AndroidRuntime(29972): 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1133)
E/AndroidRuntime(29972): 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:607)
E/AndroidRuntime(29972): 	at java.lang.Thread.run(Thread.java:762)
D/ViewRootImpl@63dfc94[MainActivity](29972): MSG_WINDOW_FOCUS_CHANGED 0
D/SurfaceView(29972): Relayout returned: oldFrame=[0,0][1440,2560] newFrame=[0,0][1440,2560] result=0x5 surface={Surface(name=null)/@0x6d52cf5 isValid=false 0}
D/ViewRootImpl@63dfc94[MainActivity](29972): mHardwareRenderer.destroy()#1
D/ViewRootImpl@63dfc94[MainActivity](29972): Relayout returned: oldFrame=[0,0][1440,2560] newFrame=[0,0][1440,2560] result=0x5 surface={isValid=false 0} surfaceGenerationChanged=true
D/ViewRootImpl@63dfc94[MainActivity](29972): mHardwareRenderer.destroy()#4
D/ViewRootImpl@63dfc94[MainActivity](29972): dispatchDetachedFromWindow
D/InputTransport(29972): Input channel destroyed: fd=84
I/System.out(29972): (HTTPLog)-Static: isSBSettingEnabled false
I/System.out(29972): (HTTPLog)-Static: isSBSettingEnabled false
```

Flutter doctor

[✓] Flutter (Channel stable, v1.7.8+hotfix.4, on Linux, locale en_US.UTF-8)
    • Flutter version 1.7.8+hotfix.4 at /home/gilles/Documents/coding/applications/flutter
    • Framework revision 20e59316b8 (5 weeks ago), 2019-07-18 20:04:33 -0700
    • Engine revision fee001c93f
    • Dart version 2.4.0

[✓] Android toolchain - develop for Android devices (Android SDK version 28.0.3)
    • Android SDK at /home/gilles/Android/Sdk
    • Android NDK location not configured (optional; useful for native profiling support)
    • Platform android-28, build-tools 28.0.3
    • ANDROID_HOME = /home/gilles/Android/Sdk
    • Java binary at: /home/gilles/Documents/coding/applications/android-studio/jre/bin/java
    • Java version OpenJDK Runtime Environment (build 1.8.0_152-release-1343-b16-5323222)
    • All Android licenses accepted.

[✓] Android Studio (version 3.4)
    • Android Studio at /home/gilles/Documents/coding/applications/android-studio
    • Flutter plugin version 36.1.1
    • Dart plugin version 183.6270
    • Java version OpenJDK Runtime Environment (build 1.8.0_152-release-1343-b16-5323222)

[✓] VS Code (version 1.37.1)
    • VS Code at /usr/share/code
    • Flutter extension version 3.3.0

[✓] Connected device (1 available)
    • SM G920F • 04157df4d4608f1b • android-arm64 • Android 7.0 (API 24)

gillestasse, 2019-08-20:
I fixed the bug by add the method from 0.2.1 to the BluetoothConnection.dart

```
  static Future<BluetoothConnection> toAddress(String address) async {
    // Sorry for pseudo-factory, but `factory` keyword disallows `Future`.
    return BluetoothConnection._consumeConnectionID(
      await FlutterBluetoothSerial._methodChannel.invokeMethod('connect', {"address": address})
    );
  }
```

closed 2019-08-20 by gillestasse

PsychoXIVI, 2019-08-20:
But it was already here!