edufolly/flutter_bluetooth_serial #64: Method not found: 'BluetoothConnection.toAddress

gillestasse, 2019-08-20:
I am using version 0.2.2

```
Compiler message:
file:///Documents/coding/applications/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_bluetooth_serial-0.2.2/lib/FlutterBluetoothSerial.dart:245:52: Error: Method not found: 'BluetoothConnection.toAddress'.
    _defaultConnection = await BluetoothConnection.toAddress(address);
                                                   ^^^^^^^^^
lib/pages/tagReader/devicesList.dart:56:65: Error: Method not found: 'BluetoothConnection.toAddress'.
     BluetoothConnection connection = await BluetoothConnection.toAddress(device.address);
                                                                ^^^^^^^^^
Compiler failed on /home/gilles/Documents/coding/rusticity/APPFENGUR/FLUTTER/appfengur/lib/main.dart
Gradle task assembleDebug failed with exit code 1
Exited (sigterm)
```

gillestasse, 2019-08-20:
I created a pull request. Should i close the case ?

PsychoXIVI, 2019-08-20:
Buddy, you might messed up your repository clone... The function exists all the time - at least for me (checked both my offline copy and remote copy (via GitHub). 

[image]

Please double check everything. 🤔

gillestasse, 2019-08-20:
I installed 0.2.2 by updating dependency to `pubspec.yaml` and check the files locally.

https://drive.google.com/file/d/1ljEzFhOjZnf66_8fufnZ0kt5Uzn_tedh/view?usp=sharing

PsychoXIVI, 2019-08-20:
And there is it on line no 61, isn't it?

gillestasse, 2019-08-20:
You are absolutely right. 
I updated from 0.0.5 to 0.2.2
It got me confused I guess...
I did not have it locally, so I got an error.
Well sorry for the inconvenience, I'll delete my pull request.

PsychoXIVI, 2019-08-20:
I will close related issues too. Please feel free to write if necessary.

closed 2019-08-20 by PsychoXIVI