edufolly/flutter_bluetooth_serial #69: Feature Request: Compatibility with windows/linux/mac desktop

hiniesta, 2019-09-06:
I'm trying the  Flutter Desktop Embedding (https://github.com/google/flutter-desktop-embedding). I want to migrate a Android project to linux desktop. It will be nice to have compatibility of this plugin with the desktop mode.

A test shows:

> Unhandled Exception: MissingPluginException(No implementation found for method getBondedDevices on channel flutter_bluetooth_serial/methods)

husamhamu, 2020-12-05:
There is an example for Windows API: https://github.com/microsoft/Windows-universal-samples/tree/master/Samples/BluetoothRfcommChat

CodeDoctorDE, 2021-07-05:
Maybe this can be the next steps for the library: @nightscape ? Flutter is a language for cross-platform development, and it would be cool to use it on the windows, linux or mac device. Many laptops have bluetooth and can be used for this.

nightscape, 2021-07-05:
Hey @CodeDoctorDE, I don't have any plans in that direction.
My project is Android only and I can only invest time if it helps that case.
I'm open to reviewing PRs and getting stuff merged if anybody wants to put in the effort.