edufolly/flutter_bluetooth_serial #70: Missing and mismatched data while reading.

sanjay4one, 2019-09-10:
I am working in an Application where it needs to read large amount data over Bluetooth SPP. When the app reads the data, it's actually not receiving all the data as expected. At first I thought it might be the issue with the device. But I tested the same functionality with a native Android app and it works without any issue. 

I looked into the plugin source code and I added a log printing the buffer inside the onRead() function from the FlutterBluetoothSerialPlugin class. Surprisingly it receives the expected data from the device but at the flutter end the exact data is not getting send. The issue might be exist somewhere while sending the data from native code to flutter using Event Channel.

Steps:
1. Data received in the onRead() function:
2, 4, 91, 0, 0, 0, 0, 49, 57, 48, 57, 48, 51, 48, 56, 48, 48, 48, 48, 50, 53, 51, 51, 87, 65, 123, 20, 14, 64, 123, 20, 14, 64, 48, 48, 48, 48, 46, 48, 48, 48, 48, 83, 48, 48, 48, 48, 48, 46, 48, 48, 48, 48, 87, 51, 67, 3, 69, 1, 1, 49, 48, 51, 48, 48, 1, -51, -52, -20, 64, 102, 102, -54, 65, -51, -52, -84, 65, -118, 1, 0, 0, 0, 49, 57, 48, 57, 48, 51, 48, 56, 48, 48, 48, 48, 53, 48, 51, 51, 87, 65, 82, -72, 14, 64, -92, 112, 13, 64, 48, 48, 48, 48, 46, 48, 48, 48, 48, 83, 48, 48, 48, 48, 48, 46, 48, 48, 48, 48, 87, 51, 67, 3, 69, 1, 1, 49, 48, 51, 48, 48, 1, -51, -52, -20, 64, 102, 102, -54, 65, -51, -52, -84, 65, -120, 2, 0, 0, 0, 49, 57, 48, 57, 48, 51, 48, 56, 48, 48, 48, 48, 55, 53, 51, 51, 87, 65, 41, 92, 15, 64, -51, -52, 12, 64, 48, 48, 48, 48, 46, 48, 48, 48, 48, 83, 48, 48, 48, 48, 48, 46, 48, 48, 48, 48, 87, 51, 67, 3, 69, 1, 1, 49, 48, 51, 48, 48, 1, -51, -52, -20, 64, 102, 102, -54, 65, -51, -52, -84, 65, -112, 3, 0, 0, 0, 49, 57, 48, 57, 48, 51, 48, 56, 48, 48, 48, 49, 48, 48, 51, 51, 87, 65, 0, 0, 16, 64, -10, 40, 12, 64, 48, 48, 48, 48, 46, 48, 48, 48, 48, 83, 48, 48, 48, 48, 48, 46, 48, 48, 48, 48, 87, 51, 67, 3, 69, 1, 1, 49, 48, 51, 48, 48, 1, -51, -52, -20, 64, 102, 102, -54, 65, -51, -52, -84, 65, -121, 4, 0, 0, 0, 49, 57, 48, 57, 48, 51, 48, 56, 48, 48, 48, 49, 50, 53, 51, 51, 87, 65, -41, -93, 16, 64, 31, -123, 11, 64, 48, 48, 48, 48, 46, 48, 48, 48, 48, 83, 48, 48, 48, 48, 48, 46, 48, 48, 48, 48, 87, 51, 67, 3, 69, 1, 1, 49, 48, 51, 48, 48, 1, -51, -52, -20, 64, 102, 102, -54, 65, -51, -52, -84, 65, -114, 5, 0, 0, 0, 49, 57, 48, 57, 48, 51, 48, 56, 48, 48, 48, 49, 53, 48, 51, 51, 87, 65, -82, 71, 17, 64, 72, -31, 10, 64, 48, 48, 48, 48, 46, 48, 48, 48, 48, 83, 48, 48, 48, 48, 48, 46, 48, 48, 48, 48, 87, 51, 67, 3, 69, 1, 1, 49, 48, 51, 48, 48, 1, -51, -52, -20, 64, 102, 102, -54, 65, -51, -52, -84, 65, -115, 6, 0, 0, 0, 49, 57, 48, 57, 48, 51, 48, 56, 48, 48, 48, 49, 55, 53, 51, 51, 87, 65, -123, -21, 17, 64, 113, 61, 10, 64, 48, 48, 48, 48, 46, 48, 48, 48, 48, 83, 48, 48, 48, 48, 48, 46, 48, 48, 48, 48, 87, 51, 67

2. Data sent from native to flutter using EventChannel.

3. Data received in flutter connection.input.listen:
2, 4, 91, 0, 0, 0, 0, 49, 57, 48, 57, 48, 51, 48, 56, 48, 48, 48, 48, 50, 53, 51, 51, 87, 65, 123, 20, 14, 64, 123, 20, 14, 64, 48, 48, 48, 48, 46, 48, 48, 48, 48, 83, 48, 48, 48, 48, 48, 46, 48, 48, 48, 48, 87, 51, 67, 3, 69, 1, 1, 49, 48, 51, 48, 48, 1, 205, 204, 236, 64, 102, 102, 202, 65, 205, 204, 172, 65, 138, 1, 0, 0, 0, 49, 57, 48, 57, 48, 51, 48, 56, 48, 48, 48, 48, 53, 48, 51, 51, 87, 65, 82, 184, 14, 64, 164, 112, 13, 64, 48, 48, 48, 48, 46, 48, 48, 48, 48, 83, 48, 48, 48, 48, 48, 46, 48, 48, 48, 48, 87, 51, 67, 3, 69, 1, 1, 49, 48, 51, 48, 48, 1, 205, 204, 236, 64, 102, 102, 202, 65, 205, 204, 172, 65, 136, 2, 0, 0, 0, 49, 57, 48, 57, 48, 51, 48, 56, 48, 48, 48, 48, 55, 53, 51, 51, 87, 65, 41, 92, 15, 64, 205, 204, 12, 64, 48, 48, 48, 48, 46, 48, 48, 48, 48, 83, 48, 48, 48, 48, 48, 46, 48, 48, 48, 48, 87, 51, 67, 3, 69, 1, 1, 49, 48, 51, 48, 48, 1, 205, 204, 236, 64, 102, 102, 202, 65, 205, 204, 172, 65, 144, 3, 0, 0, 0, 49, 57, 48, 57, 48, 51, 48, 56, 48, 48, 48, 49, 48, 48, 51, 51, 87, 65, 0

As you can see both the data are different. 

Any help is appreciated. Thanks in advance.

PsychoXIVI, 2019-09-10:
You added native read log to `onRead` from `BluetoothConnectionWrapper` class in `FlutterBluetoothSerialPlugin` class, right? 

From the values, it looks like something went wrong in coding the data while sending to common code (Flutter side).

Could you attach here which version of Flutter are you using? Log of `flutter --version` would be excellent. 

I will setup some tests in hour.

sanjay4one, 2019-09-10:
@PsychoXIVI 
Right, onRead from BluetoothConnectionWrapper.

Flutter 1.7.8+hotfix.4 • channel stable • https://github.com/flutter/flutter.git
Framework • revision 20e59316b8 (8 weeks ago) • 2019-07-18 20:04:33 -0700
Engine • revision fee001c93f
Tools • Dart 2.4.0

Thanks for your immediate response. Really appreciate it. I have implemented the plugin in a production application and got stuck due to this issue.

sanjay4one, 2019-09-10:
> From the values, it looks like something went wrong in coding the data while sending to common 
> code (Flutter side). For Java on native side, bytes are from -128 to 127. For Dart bytes are stored in 
> Uint8List and these are from 0 to 255. There might be some issue with converting between these

Yeah. But also note that the length of both the data set are different. At-least data should not have been lost.

sanjay4one, 2019-09-10:
@PsychoXIVI , I think the length issue I am talking about is not there. Earlier I just printed the data and all the data was not getting printed on the Logcat terminal may be due to print length restriction. I just printed the length of the byte array at the native and flutter and those are same. So I think it was not a bug. About the byte value conversion, I think dart is doing as per it's defined range which is 0 to 255.

PsychoXIVI, 2019-09-10:
I prepare [some tests ](https://github.com/PsychoXIVI/flutter_bluetooth_serial/commit/4d154189870b91b3b4cb894f45878f626c32eb82) and noticed two issues:

### Either Android logging or Flutter `print` are retarded and they limits output
```dart
print('XD ' + Iterable.generate(256).join(', '));
```
outputs:
```
I/flutter ( 6573): XD 0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35, 36, 37, 38, 39, 40, 41, 42, 43, 44, 45, 46, 47, 48, 49, 50, 51, 52, 53, 54, 55, 56, 57, 58, 59, 60, 61, 62, 63, 64, 65, 66, 67, 68, 69, 70, 71, 72, 73, 74, 75, 76, 77, 78, 79, 80, 81, 82, 83, 84, 85, 86, 87, 88, 89, 90, 91, 92, 93, 
94, 95, 96, 97, 98, 99, 100, 101, 102, 103, 104, 105, 106, 107, 108, 109, 110, 111, 112, 113, 114, 115, 116, 117, 118, 119, 120, 121, 122, 123, 124, 125, 126, 127, 128, 129, 130, 131, 132, 133, 
134, 135, 136, 137, 138, 139, 140, 141, 142, 143, 144, 145, 146, 147, 148, 149, 150, 151, 152, 153, 154, 155, 156, 157, 158, 159, 160, 161, 162, 163, 164, 165, 166, 167, 168, 169, 170, 171, 172, 173, 174, 175, 176, 177, 178, 179, 180, 181, 182, 183, 184, 185, 186, 187, 188, 189, 190, 191, 192, 193, 194, 195, 196, 197, 198, 199, 200, 201, 202, 203, 204, 205, 206, 207, 208, 209, 210, 211, 212, 213, 214, 215, 216, 217, 218, 219, 220, 221, 222, 223, 224, 225,
```
See https://github.com/flutter/flutter/issues/22665 for this issue.
That is why there might be difference in lengths in outputs. Also, in my tests, both sent and received buffer lengths are the same - they just does not print properly. Please check it for yourself too by printing `receivedData.length` or something and comparing. 

### Difference in value encoding in Java and Dart for bytes
Java sends -128, Dart receives 128 - both `10000000` in binary. Standard Flutter messaging channel is using just binary to push data trough. 

I will look for proper way of passing value between these and prepare some patch.

PsychoXIVI, 2019-09-10:
In your case, you might apply additional input transformation, because your remote device is encoding the data in Java manner.

sanjay4one, 2019-09-10:
@PsychoXIVI 

> Either Android logging or Flutter print are retarded and they limits output
Yeah that was the issue. I felt myself dumb when I realized it. Anyway thank you so much for your time and efforts that you invested to look into my issue. Appreciate that.

PsychoXIVI, 2019-09-10:
No problem, feel free to ask for further help if necessary.

closed 2019-09-10 by PsychoXIVI

sanjay4one, 2019-09-10:
@PsychoXIVI Sure. I will be using this plugin in many of our projects. Thanks for developing the plugin.