edufolly/flutter_bluetooth_serial #75: Is "Access list of installed apps" permission necessary ?

cindyqin2015, 2019-09-24:
After build flutter bluetooth serail into my app, and make a release package.
I found app ask for "Access list of installed apps" permission. 
Is it necessary for bluetooth communication?  what is the purpose for this permission? 
Is possible to remove it?

Thank you.

PsychoXIVI, 2019-09-24:
This plugin does not require such permission. This must come from your application I think. Please check  your `AndroidManifest.xml`.

The plugin base `AndroidManifest` addition, which integrates with your `AndroidManifest` in APK building process, looks like:
https://github.com/edufolly/flutter_bluetooth_serial/blob/46aefe40656b3b8422aa397008439d39e073a983/android/src/main/AndroidManifest.xml#L1-L6

Details:
* `android.permission.BLUETOOTH`  - essential one to access most basic Bluetooth API on Android,
* `android.permission.BLUETOOTH_ADMIN` - allows to manage connections to devices (including pairing) and control Bluetooth adapter (like enable/disable Bluetooth from your app).
* `android.permission.ACCESS_COARSE_LOCATION` - required in order to scan/discovery other devices.

cindyqin2015, 2019-09-25:
I saw flutter_bluetooth_serial_example requires "Access list of installed apps" permission under "Set individual permissions"-> "OTHER PERMISSIONS", that's why I guess this plugin contain it.

PsychoXIVI, 2019-09-25:
Example app also does not require such permission, please take a look at its `AndroidManifest`:
https://github.com/edufolly/flutter_bluetooth_serial/blob/master/example/android/app/src/main/AndroidManifest.xml

To be sure, I also recompiled a moment ago the example app for myself using: `flutter build apk --release` and then I checked final `AndroidManifest` and it does not contain more that specified above permission - I used https://www.sisik.eu/apk-tool to do so. I also installed the app on my Android 8.0 phone to check it by copying `apk` from output to SD card and then force installing it.

SISIK' apk-tool output:
https://pdfhost.io/v/5omX6yqlB_Online_APK_Analyzer___sisikpdf.pdf

Some screenshots of installation on my Android:
https://imgur.com/a/KSNFXah

Please check your app standalone without `flutter_bluetooth_serial` plugin. If possible, please attach screenshots of permission on list, Flutter and Android version numbers you are using (`flutter --version` in console` and `adb shell "getprop | grep ro.build"` command might be useful).

cindyqin2015, 2019-09-25:
Yes, I upload my apk to https://www.sisik.eu/apk-tool, and found only 3 permissions required: 
android.permission.BLUETOOTH
android.permission.BLUETOOTH_ADMIN
android.permission.ACCESS_COARSE_LOCATION

And I also post screenshots of permissions on my phone Huawei p9 plus, Android version 8.0.0:
[https://imgchr.com/i/ueiXZj](https://imgchr.com/i/ueiXZj)
[https://imgchr.com/i/ueiLLQ](https://imgchr.com/i/ueiLLQ)

You can found "Access list of installed apps" in second link, may be it is the wrong translation for one of upper permission?

PsychoXIVI, 2019-09-25:
Indeed, something is odd. 🤔 

I was testing on my Huawei Mate 10 Lite, also 8.0 Android version. I have no "Set individual permission" opt-in, I can only manage default set of permissions (camera, location, etc.). I can only view these using button in task-bar menu (3 dots). Maybe my device just don't list these at all...

I will try to setup virtual Android device in few hour just to check - Maybe there is option to set individual permissions. Also, please attach Flutter version number you are using - I might also check specific version, since I am using latest beta at the moment.

Either it is just hidden on my device or not, I don't see what could I do about it, sorry. I don't think there is even related code to listing apps or something in whole library. 

Maybe Flutter as framework requires it? Are you sure other apps created using Flutter on your setup have no such permission required? I will also ask on some forums for advice ([/r/FlutterDev Discord ](  https://discord.gg/N7Yshp4)especially). 

And I am sure that it isn't about translation.

PsychoXIVI, 2019-09-25:
Unfortunately I cannot use emulator right now because of https://issuetracker.google.com/issues/141613548 so I cannot check that for myself. You can try test it for yourself, if so please let me know of your result - I am quite curious about this permission issue.

cindyqin2015, 2019-09-26:
My flutter version:
Flutter 1.9.1+hotfix.2 • channel stable • https://github.com/flutter/flutter.git
Framework • revision 2d2a1ffec9 (3 weeks ago) • 2019-09-06 18:39:49 -0700
Engine • revision b863200c37
Tools • Dart 2.5.0

and I check the sample app of google "counter", it is not request permission.
and some tutorial app:
https://github.com/shakib609/todos-flutter
it is not request permission too.

Bluetooth hardware relative can't run on simulator.
I will try to find another phone later today. To see whether this is Huawei only issue or not and update to you.

cindyqin2015, 2019-09-26:
I try SAMSUNG Galaxy S7, android 8.0.0. it shows different permissions:
1. access approximate location
2. have full network access
3. pair with Bluetooth devices
4. access Bluetooth settings.

So it looks like different manufactory has different explanation for permissions, any way I think we can close this question now.

closed 2019-09-26 by cindyqin2015