edufolly/flutter_bluetooth_serial #76: 中文乱码（Chinese garbled）

liangguocheng, 2019-10-08:
我给蓝牙设备发送中文字符时，蓝牙打印机出现中文乱码，字母和数字能正常打印，希望作者能为我解决问题，感谢。
When I sent Chinese characters to the bluetooth device, the bluetooth printer appeared Chinese scrambled codes. Letters and Numbers can be printed normally. I hope the author can solve the problem for me. Thank you!

PsychoXIVI, 2019-10-08:
What code did you used to sent the string to the printer? Did you used good encoding? Please share details about the printer - it may require specific encoding to print all the things properly.

PsychoXIVI, 2019-11-02:
Closing due to no response. Reminder: If you found solution to your problem you should comment with your results - these may be useful for further readers, who came across similar problem.

closed 2019-11-02 by PsychoXIVI