edufolly/flutter_bluetooth_serial #79: scan for bluetooth devices while connected

simeon9696, 2019-12-22:
Hello, I'm trying to build an app that measures the distance of a device based on the RSSI value of the connected device. I know this is not the most accurate way but it's more a challenge than anything.

On the first run, discovering and pairing with devices is fine. devices are discovered and bonded successfully. I do the discovery using:
```
  StreamSubscription<BluetoothDiscoveryResult> _streamSubscription;
  List<BluetoothDiscoveryResult> _devices = List<BluetoothDiscoveryResult>();

  _streamSubscription = FlutterBluetoothSerial.instance.startDiscovery().listen((r) {
     setState(() { _devices.add(r); });
  });
```

When I bond to the device, I save the address and navigate to another route and attempt to scan (while connected) for devices again but this time identify the one I just saved.
```
  StreamSubscription<BluetoothDiscoveryResult> _streamSubscription;
  _streamSubscription = FlutterBluetoothSerial.instance.startDiscovery().listen((r) {
     if(r.device.address == savedDeviceAddress){
          print(r.rssi);
     }
  });
```

However this  gives
```
D/FlutterBluePlugin( 8479): Starting discovery
D/FlutterBluePlugin( 8479): Canceling discovery (stream closed)
```
The stream starts and closes instantly and I'm not sure why. Since the stream closes successfully on the first scan (it does that automatically after discovery closes), I would think that I'm allowed to open a new stream to scan but I feel like I to close it somewhere but I don't know how. 

Is there anyway to start a new scan while connected or get the RSSI of a connected device using this plugin?

### Environment
flutter_bluetooth_serial: ^0.2.2

ajayrathode, 2020-04-17:
@simeon9696 Hi , I also want to do the same thing calculating rough distance from RSSI value of bluetooth signal. Seems like you are also doing the same. Would like to know if you achieved the above. Please let me know.

simeon9696, 2020-04-17:
@ajayrathode unfortunately I haven't been successful. Still waiting on a response from the authors

ajayrathode, 2020-04-18:
Oh ok. But you said you able to discover list of devices and rssi value of
them . Am i Right ?

PsychoXIVI, 2020-04-18:
How about invoking `cancel` of `StreamSubscription<BluetoothDiscoveryResult>` on `dispose` of previous page (the one with first discovering)? Maybe it isn't really closed? The example code lack it as well...

_Sorry for lack of support, but I can't test those by myself - unfortunately, I have no free time (job&college)._

PsychoXIVI, 2020-04-18:
@ajayrathode for playin discovery (including RSSI values) you might want to take look at example app. There is a discovery page.

simeon9696, 2020-04-18:
> How about invoking `cancel` of `StreamSubscription<BluetoothDiscoveryResult>` on `dispose` of previous page (the one with first discovering)? Maybe it isn't really closed? The example code lack it as well...
> 
> Whole RSSI refreshing might be also hard to do. You might have to restart discovery just every time you discovered the selected device. Whole discovery process is quite slow too - it would be annoying to work with.
> 
> Besides, I think measuring distance by RSSI is so imprecise, that is a waste of time. Every device can use stronger or weaker signal strength as it differs by hardware. I think, at most you can use it more like hint to find direction or something.
> 
> _Sorry for lack of support, but I can't test those by myself - unfortunately, I have no free time (job&college)._

Thanks for the suggestion! Yes it wasn't intended for precise measurement, just more of a guide. Totally understand the job and college bit but maybe add something to the top of the plugin readme to indicate that?