edufolly/flutter_bluetooth_serial #81: Crash after using "View background collected data"

devzom, 2020-02-21:
Hey, I experienced an issue when tried to use button => "View background collected data" after turning bluetooth, pairing to BT device and using " Connect to start background collecting"

Clicking the "View background collected data" move towards an error and during flutter debugging shows " Bad state: No element. "  and on debugger console:

`════════ Exception caught by widgets library ═══════════════════════════════════════════════════════
The following StateError was thrown building BackgroundCollectedPage(dirty, dependencies: [_InheritedModel<BackgroundCollectingTask>]):
Bad state: No element

The relevant error-causing widget was: 
  BackgroundCollectedPage file:///C:/Users/jakub.zomerfeld/Documents/GitHub/Flutter/bluetooth/flutter_bluetooth_serial-issue70_debug/example/lib/MainPage.dart:314:34
When the exception was thrown, this was the stack: 
#0      List.first (dart:core-patch/growable_array.dart:219:5)
#1      BackgroundCollectedPage.build (package:flutter_bluetooth_serial_example/BackgroundCollectedPage.dart:13:45)
#2      StatelessElement.build (package:flutter/src/widgets/framework.dart:4547:28)
#3      ComponentElement.performRebuild (package:flutter/src/widgets/framework.dart:4478:15)
#4      Element.rebuild (package:flutter/src/widgets/framework.dart:4202:5)
...
════════════════════════════════════════════════════════════════════════════════════════════════════
`

Moreover  noticed that whenever I connect to any BT device by "Connect to start background collecting" in debug console shows ` getBluetoothService() called with no BluetoothManagerCallback`

### Environment

Flutter 1.15.4-pre.117 • channel master • https://github.com/flutter/flutter.git
Framework • revision d030296a97 (27 hours ago) • 2020-02-20 01:51:15 -0800
Engine • revision f2f8c342be
Tools • Dart 2.8.0 (build 2.8.0-dev.9.0 0f141be8bd)

PsychoXIVI, 2020-02-22:
> Clicking the "View background collected data" move towards an error and during flutter debugging shows " Bad state: No element. " and on debugger console:

It looks like small async bug: `BackgroundCollectingPage ` tries to used data from `BackgroundCollectingTask ` asking for first entry. But there is no guarantee there is already first element in there - so we should further await for first sample as we starting the collecting task. 

For example I assumed the user cannot click as fast between connecting and viewing the results, that there will be no data to view. Apart of the "bug", there should be the data already.

Are you sure you were connected to the device? Was device sending the data?

> Moreover noticed that whenever I connect to any BT device by "Connect to start background collecting" in debug console shows getBluetoothService() called with no BluetoothManagerCallback

I don't think I can do something about it: https://stackoverflow.com/questions/15889908/getbluetoothservice-called-with-no-bluetoothmanagercallback

devzom, 2020-02-27:
Thank You for response. I'll try to work with the async, make some validation and delay. 

Yeap, I was connected within device.
I didn't clicked to fast. I connected it and waited even for  a  minute, but that doesn't change anything. I'm trying to rebuild this feature from stratch.  

Going to continue this project? or You think it's worth to just go through Flutter_blue?

Overall, thank You :)

closed 2020-02-27 by devzom

PsychoXIVI, 2020-02-27:
> Yeap, I was connected within device.

Might I ask, what device setup are you using? The `BackgroundCollecting` example was cut just for simple Arduino code. Maybe there is something wrong in external device program?

> Going to continue this project? or You think it's worth to just go through Flutter_blue?

I would love to continue project, but I have both work and college on my head :/  

You might want to try with `flutter_blue` package, but I don't know does it work on Bluetooth Classic/Serial (not the same as Bluetooth Low Energy).

devzom, 2020-03-17:
I know it's designed for arduino. I'm changing it to work with "MPE" low energy bluetooth device as bicycle onboard computer connected with electric engine. Working partially atm. :)

I see the point, have the same situation! Wish You a good luck on studies!

Not trying Flutter blue atm.
Thanks for support :)