edufolly/flutter_bluetooth_serial #82: Add test cases

bean5, 2020-02-27:
### Problem summary
Most flutter projects come with test cases, even if they are not great. This project seems to have 0 cases.

### Steps to reproduce
Main Project
1. Clone project
2. cd into project and run `flutter test`

Example Project
1. `cd example`
2. `flutter test`

### Environment

N/A

PsychoXIVI, 2020-02-27:
Its quite hard to create automated tests that require external hardware. There are also so much code on native side. There is no simple way to test such things in Flutter. Still, I understand concern, and see some possible solutions:
* mocking every single native coded thing,

  That could be possible, but such "mock" could be too big itself. Also, its only testing Dart code, not the native which is mocked up.

* special app and code for external device,

  Some Raspberry with script that reset its state for each tests? Both testing script on Raspberry and Flutter console could synchronize and communicate by network or something.

bean5, 2020-03-08:
I understand the problems you mention. Interfacing with actual hardware for testing is hard to trust. They'd be flaky, prone to error.

If you truly have no testable methods because all of them depend on hardware, I'd advise creating a test file with a comment indicating that. It is okay to say "Not feasible". But leaving it blank makes one wonder if the project was created with really old template, is not trustworthy, etc.

So presence of even a nearly enough fine bolsters more trust in a project in my opinion. It'll also open the doors of your mind to that test case you could write it didn't realize because you thought there weren't any to add... Or a case someone else might think to add.