edufolly/flutter_bluetooth_serial #85: Why is location access required for accessing Bluetooth?

sbis04, 2020-03-12:
### Problem summary

The app requires **location access** every time it is running on the device. I believe Bluetooth doesn't require a location permission.

### Steps to reproduce

1. Open example app
2. Give permission to turn on Bluetooth
3. Then `Allow flutter_bluetooth to access this device's location?` dialog-box appears

Removing the following line from the **AndroidManifest.xml** file:
```xml
<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION"/>
``` 

### Environment

#### `flutter --version` output from the terminal:
```
Flutter 1.12.13+hotfix.8 • channel stable •
https://github.com/flutter/flutter.git
Framework • revision 0b8abb4724 (4 weeks ago) • 2020-02-11 11:44:36 -0800
Engine • revision e1e6ced81d
Tools • Dart 2.7.0
```

PsychoXIVI, 2020-03-12:
The permission is required in order to discover (scanning for) other Bluetooth devices by the library. Well, maybe someday it could be optional - I don't think you need it, if you are connecting to fixed address...

_Why closed with no message?_

sbis04, 2020-03-12:
Sorry for closing the issue without any message. But I later found out that for discovering as well as for connecting to paired devices also, location permission is required.

But, for the end-user, this seems various suspicious as to "**Why a Bluetooth connection should require location access?**"

YusufElosman, 2020-03-15:
Is there a solution for not ask location permission ?

PsychoXIVI, 2020-03-15:
There probably is, but plugin does not YET implement it. 

As I mentioned above, the plugin does not "require" the permission, but scanning/discovering devices do. Theoretically, plugin should allow for direct connections to known devices without the permission.

It require some rewrite of platform (Java) code. Feel free to fork and implement it yourself, and then pull request here :)

yanivshaked, 2021-11-09:
PR sumitted [here](https://github.com/edufolly/flutter_bluetooth_serial/pull/152), no need to ask for location permissions if you want to get list of bonded devices.