edufolly/flutter_bluetooth_serial #87: Connection is lost upon hot-restart (and multitasking) and can never be recovered

nachoco, 2020-04-09:
### Problem summary
After creating a connection, if the app is hot-restarted the connection reference is lost and there is no way to retrieve it

### Steps to reproduce

1. getBondedDevices. 
2. Let the user choose one. Via code connect
3. Hot-Restart the app from the IDE
4. State is lost, the currentConnection reference I have is lost.
5. calling getBondedDevices tells me that my device is connected, which is true.
6. No way to get my BluetoothConnection reference again. If I try to connect() again of course I get an error.

PsychoXIVI, 2020-04-10:
> multitasking
Do it really occurs for you while multitasking? Does app lose its state? Does it go back to main page? 

> hot-restart
I think, if plugin (native/java code) isn't restarted on hot-restart, problem with retrieving the connection after the restart could be solved by adding some `getActiveConnections` which could get and somewhat parse native list of active connections (`BluetoothConnectionWrapper`s list). Those would require to recreate event channels (read sinks) too.

luccasmaso, 2020-06-17:
Or maybe flutter could have an option to register callbacks when a hot-restart occurs. Then your app code could gracefully end the current connection when this signal is sent. This way I think a new connection will work fine. 

I've found some other related problems with socket based connection libs around the community:

https://github.com/flutter/flutter/issues/27463
https://github.com/flutter/flutter/issues/10437
https://github.com/flutter/flutter/issues/48586

luccasmaso, 2020-06-17:
@PsychoXIVI Did you find out that `getActiveConnections` is possible?