edufolly/flutter_bluetooth_serial #92: How do we print a test with connected printers

jjoao07, 2020-05-29:
Hi there,

I have followed the instructions of this plugin, but I can't see a way to print data or send data from the app to the printer.

can you guys please help out how do we print a simple image file or a QR code from the app to a Bluetooth connected printer.

Thanks

edufolly, 2020-05-29:
Hi João,

you need to use the printer language to communicate. Look at this issue #1 

Best.

jjoao07, 2020-05-29:
Hi @edufolly thanks for your reply.
My printer is Zebra, Qln320, the function or method I’m looking on is how to call something like `bluethoothPrinter.print()`
Just to send an Hello word to connected printer.

edufolly, 2020-05-29:
I never integrate this plugin with a BT printer.

I find this interesting information on Zebra's site.
https://www.zebra.com/us/en/support-downloads/printer-software/by-request-software.html

Maybe you need ZPL to print your tag. Hope this helps

jjoao07, 2020-05-30:
Oi @edufolly muito obrigado ai pela ajuda,
acabei descobrindo que a para fazer a impressao do QR code com essa impressora,
tenho de escrever o ZPL comandos.

so vou tou revisando o manual todo na pagina do Zebra, qualquer coisa dou um sinal.
Bom dia

closed 2021-07-16 by edufolly