edufolly/flutter_bluetooth_serial #93: requestDisable() user interaction

iaexc, 2020-06-02:
A dialog appears with message "this app wants to disable bluetooth..." on one of my test devices (Huawei brand), while this doesn't appear in my other devices. Can I get this library to not show a dialog to ask user to disable bluetooth in any devices?

PsychoXIVI, 2020-06-03:
I don't think so. It may be related to how your OS handles permissions. This may be passed when appliacation uses special system-only permissions (BLUETOOTH_PRIVILEGED in this case), but those require app to be properly signed. Of course, this permission is not added to library manifest, because it marks unproperly signed or unsigned apps (including example) as invalid or adds warning before instaling those (the permission is still discarded tho)

iaexc, 2020-06-04:
Thanks for the info!

closed 2020-06-04 by iaexc