edufolly/flutter_bluetooth_serial #98: Discovery not working with sdk version 29 on certain devices

NailAgliev, 2020-07-23:
### Problem summary

Call of `startDiscovery()` just produce those lines and nothing else.

    D/FlutterBluePlugin(21036): Starting discovery
    D/BluetoothAdapter(21036): startDiscovery

With SDK version 28 works as expected.

### Steps to reproduce

Issue found only during testing on Samsung Galaxy S10 plus SM G975U1

    import 'package:flutter/material.dart';
    import 'package:flutter_bluetooth_serial/flutter_bluetooth_serial.dart';

    void main() async {
      WidgetsFlutterBinding.ensureInitialized();

      FlutterBluetoothSerial.instance.startDiscovery().listen((result) {
        print(result.device.address);
      });
    }

PsychoXIVI, 2020-08-30:
> If your app targets Android 10 or higher, it must have the ACCESS_FINE_LOCATION permission in order to use several methods within the Wi-Fi, Wi-Fi Aware, or Bluetooth APIs.

https://developer.android.com/about/versions/10/privacy/changes

Maybe adding the `ACCESS_FINE_LOCATION ` permission will help it work on SDK 29?

I don't have how to test it myself.

magnuswikhog, 2021-02-05:
This pull request seems to fix the issue of not working on Android 10, https://github.com/edufolly/flutter_bluetooth_serial/pull/110

Hopefully the authors will merge the pull request one day... until then, I guess we have to clone the plugin repo and make necessary changes ourselves, which is a shame on an otherwise very useful plugin.

hainv23, 2021-05-11:
Add ACCESS_FINE_LOCATION permission worked for me

qilanjie, 2021-05-29:
add ACCESS_BACKGROUND_LOCATION