edufolly/flutter_bluetooth_serial #101: Flutter_Bluetooth_Serial getBluetoothService() called with no BluetoothManagerCallback Error

Amin-Matola, 2020-09-04:
I am trying to connect to bluetooth device using flutter bluetooth serial module, but everytime I get error as:

```shell
C:\Users\Me>...
D/ViewRootImpl(16512): ViewPostImeInputStage processPointer 1
D/FlutterBluePlugin(16512): Starting discovery
D/BluetoothAdapter(16512): startDiscovery
D/BluetoothAdapter(16512): startDiscovery = true
D/FlutterBluePlugin(16512): Discovered B8:27:EB:C9:45:29
D/FlutterBluePlugin(16512): Discovery finished
D/BluetoothAdapter(16512): cancelDiscovery
D/BluetoothAdapter(16512): cancelDiscovery = true
D/FlutterBluePlugin(16512): Canceling discovery (stream closed)
D/BluetoothAdapter(16512): cancelDiscovery
D/BluetoothAdapter(16512): cancelDiscovery = true
D/ViewRootImpl(16512): ViewPostImeInputStage processPointer 0
D/ViewRootImpl(16512): ViewPostImeInputStage processPointer 1
D/FlutterBluePlugin(16512): Connecting to B8:27:EB:C9:45:29 (id: 3)
D/BluetoothAdapter(16512): cancelDiscovery
D/BluetoothAdapter(16512): cancelDiscovery = true
D/BluetoothUtils(16512): isSocketAllowedBySecurityPolicy start : device null
D/BluetoothSocket(16512): connect(): myUserId = 0
W/BluetoothAdapter(16512): getBluetoothService() called with no BluetoothManagerCallback
E/System  (16512): Uncaught exception thrown by finalizer
E/System  (16512): java.io.IOException: socket not created
E/System  (16512): 	at android.net.LocalSocketImpl.shutdownInput(LocalSocketImpl.java:396)
E/System  (16512): 	at android.net.LocalSocket.shutdownInput(LocalSocket.java:206)
E/System  (16512): 	at android.bluetooth.BluetoothSocket.close(BluetoothSocket.java:793)
E/System  (16512): 	at android.bluetooth.BluetoothSocket.finalize(BluetoothSocket.java:308)
E/System  (16512): 	at java.lang.Daemons$FinalizerDaemon.doFinalize(Daemons.java:217)
E/System  (16512): 	at java.lang.Daemons$FinalizerDaemon.run(Daemons.java:200)
E/System  (16512): 	at java.lang.Thread.run(Thread.java:818)
E/System  (16512): Uncaught exception thrown by finalizer
E/System  (16512): java.io.IOException: socket not created
E/System  (16512): 	at android.net.LocalSocketImpl.shutdownInput(LocalSocketImpl.java:396)
E/System  (16512): 	at android.net.LocalSocket.shutdownInput(LocalSocket.java:206)
E/System  (16512): 	at android.bluetooth.BluetoothSocket.close(BluetoothSocket.java:793)
E/System  (16512): 	at android.bluetooth.BluetoothSocket.finalize(BluetoothSocket.java:308)
E/System  (16512): 	at java.lang.Daemons$FinalizerDaemon.doFinalize(Daemons.java:217)
E/System  (16512): 	at java.lang.Daemons$FinalizerDaemon.run(Daemons.java:200)
E/System  (16512): 	at java.lang.Thread.run(Thread.java:818)
```
____________________________________________________________________________________________________________________________
My code to connect the device is...

```dart
Future<void> connect(address) async {

    BluetoothConnection.toAddress(address).then((conn){
      print("Connected to device $address");
      conn.input.listen((Uint8List event) {
        print(event);
      }).onDone(() {
        print("Closing connection");
        conn.close();
      });
      showDialog(context: context, builder: (d){
        return AlertDialog(
          title: Text("Connection Successful"),
          content: Text("Connected to device $address"),
          actions: [
            FlatButton(
              child: Text("Ok"),
              onPressed: (){
                Navigator.pop(context);
              },
            )
          ],
        );
      });
    })
    .catchError((error){
          showDialog(context: context, builder: (d){
            return AlertDialog(
              title: Text("Connection Problem!", style: TextStyle(color: Colors.red,),),
              content: Text("There was an error connecting to device $address", style: TextStyle(color: Colors.grey[600]),),
              actions: [
                FlatButton(
                  child: Text("Ok"),
                  onPressed: (){
                    Navigator.pop(context);
                  },
                )
              ],
            );
          });
        });
  }
```

PsychoXIVI, 2020-10-16:
What Android version are you using?

Amin-Matola, 2020-10-19:
> What Android version are you using?

Well, I ran the app on both "Android 6 and 7", all to notice the same!

umamakhalid, 2020-10-21:
When will you resolve this issue. I need to connect my BT headsets for live audio transmission. but on connection I get this error.

patzu, 2021-01-08:
Could you solve this problem? I also face with this error. I run Android app on two android device and they cannot connect to chat.

umamakhalid, 2021-01-08:
No sorry. I have stop using flutter as there was no fine solution to this
problem on internet.

devfelipereis, 2021-02-01:
Same here. Using android 10.

Luffys6, 2021-09-03:
Same here. Using android 8.0.

campanagerald, 2021-12-10:
Has anyone solved this problem?