edufolly/flutter_bluetooth_serial #105: Explore discovered devices i.e (FlutterBluetoothSerial.instance.startDiscovery()) does not work on Samsung devices while they work for Oppo mobile phones.

umamakhalid, 2020-10-21:
### Problem summary

### Steps to reproduce

1. 
2. 
3. 
4. 

### Environment

```
```

changhouw, 2020-10-29:
We have same issue, try to request permission fine location before discovering

vaibhiarora03, 2021-04-07:
I am requesting fine location permissions but still the same issue. The devices are discovered once I switch from phone's bluetooth settings.

umamakhalid, 2021-04-07:
Try giving background location permission

vaibhiarora03, 2021-04-07:
Tried but same issue.

<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION"/>
<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION"/>
<uses-permission android:name="android.permission.ACCESS_BACKGROUND_LOCATION"/>

umamakhalid, 2021-04-07:
Try asking permissions inside the code. Permission for background location
and other bluetooth locations.

umamakhalid, 2021-04-16:
1. Give Background Location Permission in Manifest
2. Few mobiles does not work with manifest permission so you also have to ask for permission in the code.

closed 2021-04-16 by umamakhalid