edufolly/flutter_bluetooth_serial #106: Can't chat with near by device

PedroRTomas, 2020-10-30:
Hello, 
I don't know if this is normal, either because it hasn't been yet implemented or because it's simply not the purpose of this package, however, if I have 2 phones (both running the same app), shouldn't I be able to use this feature?
Here is the error generated by each one: (Which seems to be the same)
[image]

In case this is not supposed to work, what is necessary that the other device runs in order to allow this chat feature? For instance, let's say that the other device is a laptop. Some script made using python, preparing the computer to be listening at a specified port for incoming Bluetooth connections would allow this type of system to work?
Something like this: 
[image]
Origin: https://blog.kevindoran.co/bluetooth-programming-with-python-3/

Thanks in advance!

PsychoXIVI, 2020-11-04:
Currently, there is no way to connect 2 phones directly using this package - it could be done, by implementing server.

As for indirect approach with some server like you trying to use, I think it would be possible, but you are missing few things:

+ Server should be registered to Service Discovery Protocol. Here some bash commands to explore:
    ```bash
    sudo sdptool add SP         # There can be channel specified one of 79 channels by adding `--channel N`.
    $ sudo sdptool browse local   # Check on which channel RFCOMM will be operating, to select in next step.
    ```
    I don't know if can you add 2 ports as Serial Port (SP), never tested it by myself. The library uses single UUID to find port in SDP, so when 2 devices are about to connect, I can only guess it will take second port if first is already taken.

+ You should run 2 servers, or rather: be able to accept 2 connection the same time. 

    The script could work for one connection only. I think you could run 2 of those in background (tho, with different port selected); then using some named pipes you could make them communicate. I don't like "multi-threading" in Python, so I can't really help with those scripts.

+ If the same message persist on mobile/client devices, there might be various of other reasons: see my other old comment: https://github.com/edufolly/flutter_bluetooth_serial/issues/18#issuecomment-506472861

PedroRTomas, 2020-11-06:
@PsychoXIVI Thank you for your answer!
====== Partial Solution ======
Eventually, I found a very basic way of doing part of what I intended, following your tips about Serial Ports.
For future reference, what I've done was:
(In one terminal)
sudo sdptool add SP --channel=22 SP
sudo rfcomm listen /dev/rfcomm0 22
(In another terminal)
sudo cat /dev/rfcomm0
===== Problem Found =====
However, while trying with two ports I've run into another problem.
While selecting another port, besides 22, the Flutter code throws a PlatformException:
[image]
I think this is the general exception thrown every time that a connection can't be established.
Now the real question is: why?
How is this package internally trying to connect? Is correct to assume that it's using port 22 as default and so every time that this port isn't available at the target machine the connection will fail?

edufolly, 2020-12-04:
Olá Pedro,

if you have further questions, please reopen this issue or open a new one.

Best.

closed 2020-12-04 by edufolly