edufolly/flutter_bluetooth_serial #108: How to connect to already connected device by the same app from another session?

fedotxxl, 2020-11-15:
Here is an explanation:

1. Connect to not paired device:
```
    BluetoothConnection connection = await BluetoothConnection.toAddress(result.device.address);
```

2. Request/push some data
3. Disconnect from device:

```
 connection.finish();
```

4. Once again try to connect to the device:
```
    BluetoothConnection connection = await BluetoothConnection.toAddress(result.device.address);
```

On step 4 app throws an exception:
```
I/flutter ( 8179): Cannot connect, exception occured PlatformException(connect_error, read failed, socket might closed or timeout, read ret: -1, java.io.IOException: read failed, socket might closed or timeout, read ret: -1
I/flutter ( 8179): 	at android.bluetooth.BluetoothSocket.readAll(BluetoothSocket.java:738)
I/flutter ( 8179): 	at android.bluetooth.BluetoothSocket.waitSocketSignal(BluetoothSocket.java:697)
I/flutter ( 8179): 	at android.bluetooth.BluetoothSocket.connect(BluetoothSocket.java:379)
I/flutter ( 8179): 	at io.github.edufolly.flutterbluetoothserial.BluetoothConnection.connect(BluetoothConnection.java:57)
I/flutter ( 8179): 	at io.github.edufolly.flutterbluetoothserial.BluetoothConnection.connect(BluetoothConnection.java:64)
I/flutter ( 8179): 	at io.github.edufolly.flutterbluetoothserial.FlutterBluetoothSerialPlugin.lambda$onMethodCall$0$FlutterBluetoothSerialPlugin(FlutterBluetoothSerialPlugin.java:880)
```

How can I use this plugin? I can't ask the user to manually unpair the device.

edufolly, 2020-12-04:
Is this possible? 
ping @PsychoXIVI

JaveedIshaq, 2021-03-06:
Is this possible?
ping @PsychoXIVI

sucper, 2021-05-19:
@edufolly 
same issue
But there is no pattern.
Sometimes it connects on the first try.
Sometimes from the second or third, but it happens that for a long time it is not possible to connect to the device (I use the HC-05 Adapter)

I am programmatically trying to connect 3 times if I get this exception. But this is a trick, not a solution. And it may not always help.

have any ideas how to fix this error?