edufolly/flutter_bluetooth_serial #119: I get the error when want chat with other device

barangungor, 2021-02-22:
Hey , i want add bluetooth chat to my app but I get this error when I click on the device I want to chat.

[Error - 12:34:58] Request textDocument/codeAction failed.
  Message: An error occurred while handling textDocument/codeAction request
  Code: -32001 
[Error - 12:34:58] An error occurred while handling textDocument/codeAction request: NoSuchMethodError: The method 'packagesAvailableTo' was called on null.
Receiver: null
Tried calling: packagesAvailableTo("c:\\src\\flutter\\.pub-cache\\hosted\\pub.dartlang.org\\flutter_bluetooth_serial-0.2.2\\lib\\FlutterBluetoothSerial.dart")
#0      Object.noSuchMethod (dart:core-patch/object_patch.dart:51:5)
#1      TransformSetManager.forLibrary (package:analysis_server/src/services/correction/fix/data_driven/transform_set_manager.dart:28:30)
#2      DataDriven._availableTransformSetsForLibrary (package:analysis_server/src/services/correction/dart/data_driven.dart:71:41)
#3      DataDriven.producers.<anonymous closure> (package:analysis_server/src/services/correction/dart/data_driven.dart:31:21)
#4      _SyncIterator.moveNext (dart:core-patch/core_patch.dart:165:25)
#5      FixProcessor._addFromProducers (package:analysis_server/src/services/correction/fix_internal.dart:1137:46)
<asynchronous suspension>
#6      FixProcessor.compute (package:analysis_server/src/services/correction/fix_internal.dart:1071:11)
#7      DartFixContributor.computeFixes (package:analysis_server/src/services/correction/fix_internal.dart:174:35)
#8      CodeActionHandler._getFixActions (package:analysis_server/src/lsp/handlers/handler_code_actions.dart:244:46)
#9      CodeActionHandler._getCodeActions (package:analysis_server/src/lsp/handlers/handler_code_actions.dart:205:7)
#10     CodeActionHandler.handle.<anonymous closure>.<anonymous closure>.<anonymous closure> (package:analysis_server/src/lsp/handlers/handler_code_actions.dart:73:18)
#11     ErrorOr.mapResult (package:analysis_server/lsp_protocol/protocol_special.dart:255:12)
#12     CodeActionHandler.handle.<anonymous closure>.<anonymous closure> (package:analysis_server/src/lsp/handlers/handler_code_actions.dart:70:26)
#13     ErrorOr.mapResult (package:analysis_server/lsp_protocol/protocol_special.dart:255:12)
#14     CodeActionHandler.handle.<anonymous closure> (package:analysis_server/src/lsp/handlers/handler_code_actions.dart:69:26)
#15     ErrorOr.mapResult (package:analysis_server/lsp_protocol/protocol_special.dart:255:12)
#16     CodeActionHandler.handle (package:analysis_server/src/lsp/handlers/handler_code_actions.dart:66:17)
#17     _rootRunUnary (dart:async/zone.dart:1198:47)
#18     _CustomZone.runUnary (dart:async/zone.dart:1100:19)
#19     _FutureListener.handleValue (dart:async/future_impl.dart:143:18)
#20     Future._propagateToListeners.handleValueCallback (dart:async/future_impl.dart:696:45)
#21     Future._propagateToListeners (dart:async/future_impl.dart:725:32)
#22     Future._completeWithValue (dart:async/future_impl.dart:529:5)
#23     _AsyncAwaitCompleter.complete (dart:async-patch/async_patch.dart:40:15)
#24     _completeOnAsyncReturn (dart:async-patch/async_patch.dart:311:13)
#25     Handler.requireResolvedUnit (package:analysis_server/src/lsp/handlers/handlers.dart)
#26     _rootRunUnary (dart:async/zone.dart:1198:47)
#27     _CustomZone.runUnary (dart:async/zone.dart:1100:19)
#28     _FutureListener.handleValue (dart:async/future_impl.dart:143:18)
#29     Future._propagateToListeners.handleValueCallback (dart:async/future_impl.dart:696:45)
#30     Future._propagateToListeners (dart:async/future_impl.dart:725:32)
#31     Future._completeWithValue (dart:async/future_impl.dart:529:5)
#32     Future._asyncCompleteWithValue.<anonymous closure> (dart:async/future_impl.dart:567:7)
#33     _rootRun (dart:async/zone.dart:1190:13)
#34     _CustomZone.run (dart:async/zone.dart:1093:19)
#35     _CustomZone.runGuarded (dart:async/zone.dart:997:7)
#36     _CustomZone.bindCallbackGuarded.<anonymous closure> (dart:async/zone.dart:1037:23)
#37     _microtaskLoop (dart:async/schedule_microtask.dart:41:21)
#38     _startMicrotaskLoop (dart:async/schedule_microtask.dart:50:5)
#39     _runPendingImmediateCallback (dart:isolate-patch/isolate_patch.dart:118:13)
#40     _Timer._runTimers (dart:isolate-patch/timer_impl.dart:404:11)
#41     _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:168:12)

[Error - 12:38:09] Request textDocument/codeAction failed.
  Message: An error occurred while handling textDocument/codeAction request
  Code: -32001 
[Error - 12:38:09] An error occurred while handling textDocument/codeAction request: NoSuchMethodError: The method 'packagesAvailableTo' was called on null.
Receiver: null
Tried calling: packagesAvailableTo("c:\\src\\flutter\\.pub-cache\\hosted\\pub.dartlang.org\\flutter_bluetooth_serial-0.2.2\\lib\\BluetoothConnection.dart")
#0      Object.noSuchMethod (dart:core-patch/object_patch.dart:51:5)
#1      TransformSetManager.forLibrary (package:analysis_server/src/services/correction/fix/data_driven/transform_set_manager.dart:28:30)
#2      DataDriven._availableTransformSetsForLibrary (package:analysis_server/src/services/correction/dart/data_driven.dart:71:41)
#3      DataDriven.producers.<anonymous closure> (package:analysis_server/src/services/correction/dart/data_driven.dart:31:21)
#4      _SyncIterator.moveNext (dart:core-patch/core_patch.dart:165:25)
#5      FixProcessor._addFromProducers (package:analysis_server/src/services/correction/fix_internal.dart:1137:46)
<asynchronous suspension>
#6      FixProcessor.compute (package:analysis_server/src/services/correction/fix_internal.dart:1071:11)
#7      DartFixContributor.computeFixes (package:analysis_server/src/services/correction/fix_internal.dart:174:35)
#8      CodeActionHandler._getFixActions (package:analysis_server/src/lsp/handlers/handler_code_actions.dart:244:46)
#9      CodeActionHandler._getCodeActions (package:analysis_server/src/lsp/handlers/handler_code_actions.dart:205:7)
#10     CodeActionHandler.handle.<anonymous closure>.<anonymous closure>.<anonymous closure> (package:analysis_server/src/lsp/handlers/handler_code_actions.dart:73:18)
#11     ErrorOr.mapResult (package:analysis_server/lsp_protocol/protocol_special.dart:255:12)
#12     CodeActionHandler.handle.<anonymous closure>.<anonymous closure> (package:analysis_server/src/lsp/handlers/handler_code_actions.dart:70:26)
#13     ErrorOr.mapResult (package:analysis_server/lsp_protocol/protocol_special.dart:255:12)
#14     CodeActionHandler.handle.<anonymous closure> (package:analysis_server/src/lsp/handlers/handler_code_actions.dart:69:26)
#15     ErrorOr.mapResult (package:analysis_server/lsp_protocol/protocol_special.dart:255:12)
#16     CodeActionHandler.handle (package:analysis_server/src/lsp/handlers/handler_code_actions.dart:66:17)
#17     _rootRunUnary (dart:async/zone.dart:1198:47)
#18     _CustomZone.runUnary (dart:async/zone.dart:1100:19)
#19     _FutureListener.handleValue (dart:async/future_impl.dart:143:18)
#20     Future._propagateToListeners.handleValueCallback (dart:async/future_impl.dart:696:45)
#21     Future._propagateToListeners (dart:async/future_impl.dart:725:32)
#22     Future._completeWithValue (dart:async/future_impl.dart:529:5)
#23     _AsyncAwaitCompleter.complete (dart:async-patch/async_patch.dart:40:15)
#24     _completeOnAsyncReturn (dart:async-patch/async_patch.dart:311:13)
#25     Handler.requireResolvedUnit (package:analysis_server/src/lsp/handlers/handlers.dart)
#26     _rootRunUnary (dart:async/zone.dart:1198:47)
#27     _CustomZone.runUnary (dart:async/zone.dart:1100:19)
#28     _FutureListener.handleValue (dart:async/future_impl.dart:143:18)
#29     Future._propagateToListeners.handleValueCallback (dart:async/future_impl.dart:696:45)
#30     Future._propagateToListeners (dart:async/future_impl.dart:725:32)
#31     Future._completeWithValue (dart:async/future_impl.dart:529:5)
#32     Future._asyncCompleteWithValue.<anonymous closure> (dart:async/future_impl.dart:567:7)
#33     _rootRun (dart:async/zone.dart:1190:13)
#34     _CustomZone.run (dart:async/zone.dart:1093:19)
#35     _CustomZone.runGuarded (dart:async/zone.dart:997:7)
#36     _CustomZone.bindCallbackGuarded.<anonymous closure> (dart:async/zone.dart:1037:23)
#37     _microtaskLoop (dart:async/schedule_microtask.dart:41:21)
#38     _startMicrotaskLoop (dart:async/schedule_microtask.dart:50:5)
#39     _runPendingImmediateCallback (dart:isolate-patch/isolate_patch.dart:118:13)
#40     _Timer._runTimers (dart:isolate-patch/timer_impl.dart:404:11)
#41     _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:168:12)

barangungor, 2021-02-22:
I found it impossible to chat between 2 phones with your plugin. I'll try with `flutter_blue`

closed 2021-02-22 by barangungor

TAMILSELVAN943, 2021-11-08:
Hi  barangungor when i use flutter_blue,when i scan no devices to display.That only reason i use this package but when i use chat that time face this issue