edufolly/flutter_bluetooth_serial #121: Migrate to null safety

savs90, 2021-03-04:
Hey guys. This is not exactly bug , but with flutter 2 announced it would be great if you can upgrade the project to null safety. Null safety is now part of the stable releases. Thanks a lot for the great work!

JAICHANGPARK, 2021-04-01:
Null Safety~~!!!

rus0n, 2021-04-18:
any new?

vishnuagbly, 2021-04-24:
pull request have been made in #122, but it looks like this project has been abandoned

CodeDoctorDE, 2021-04-27:
can anyone create a new fork which accepts pull requests?

CodeDoctorDE, 2021-05-06:
What's the new plan? @vishnuagbly

ShinyWrench, 2021-06-23:
If you just want to get the package working with your program, here's a couple ways to do it.

If you're using VS Code and the Dart and Flutter extensions, you can add this to your .vscode/settings.json (or create it in your project folder with this content):
```
{
    "dart.flutterRunAdditionalArgs": [
        "--no-sound-null-safety"
    ],
    "dart.vmAdditionalArgs": [
        "--no-sound-null-safety"
    ]
}
```
If you don't want to do it that way, you can run your program with this:
`flutter run --no-sound-null-safety`

(https://dart.dev/null-safety/unsound-null-safety#testing-or-running-mixed-version-programs)

nightscape, 2021-07-04:
Corresponding PR is merged. You can try it by using the Git version of this repo.

closed 2021-07-04 by nightscape

edufolly, 2021-07-05:
Version 0.3.0 was published.