edufolly/flutter_bluetooth_serial #123: reading data through bluetooth

goldencoder20, 2021-03-08:
Hi
currently I'm working in a small app connected with a device through bluetooth.
so sending data works fine for me using this function:
void _sendMessage() async {
    connection.output.add(utf8.encode("data here" + "\r\n"));
    await connection.output.allSent;
  }

but I don't know how to read data and print it for example in console
please can you provide a function can solve the issue?
thanks

MojamojaK, 2021-04-08:
Follow the example

This sets the callback to be called every time data is received
```Dart
connection.input.listen(_onDataReceived).onDone(() {
  if (isDisconnecting) {
    print('Disconnecting locally!');
  } else {
    print('Disconnected remotely!');
  }
  if (this.mounted) {
    setState(() {});
  }
});
```
where the callback `_onDataReceived ` is implemented like this
```Dart
void _onDataReceived(Uint8List packet) {
  try {
    print(utf8.decode(packet.sublist(0, packet.length)));
  } catch (e) {
    setState(() {});
  }
}
```

goldencoder20, 2021-04-10:
thanks a lot, it works fine now

closed 2021-04-10 by goldencoder20

heqikang, 2021-04-14:
> Follow the example
> 
> This sets the callback to be called every time data is received
> 
> ```dart
> connection.input.listen(_onDataReceived).onDone(() {
>   if (isDisconnecting) {
>     print('Disconnecting locally!');
>   } else {
>     print('Disconnected remotely!');
>   }
>   if (this.mounted) {
>     setState(() {});
>   }
> });
> ```
> 
> where the callback `_onDataReceived ` is implemented like this
> 
> ```dart
> void _onDataReceived(Uint8List packet) {
>   try {
>     print(utf8.decode(packet));
>   } catch (e) {
>     setState(() {});
>   }
> }
> ```

I sending data works fine, but read data failed

[image]
[image]