edufolly/flutter_bluetooth_serial #127: Please, migrate to null-safety

selmaskopljakovic, 2021-05-04:
Hello authors,

can you please migrate flutter_bluetooth_serila to null-safety?

Thank you!

ShinyWrench, 2021-06-23:
If you just want to get the package working with your program, here's a couple ways to do it.

If you're using VS Code and the Dart and Flutter extensions, you can add this to your .vscode/settings.json (or create that file with this content):
```
{
    "dart.flutterRunAdditionalArgs": [
        "--no-sound-null-safety"
    ],
    "dart.vmAdditionalArgs": [
        "--no-sound-null-safety"
    ]
}
```
If you don't want to do it that way, you can run your program with this:
`flutter run --no-sound-null-safety`

(https://dart.dev/null-safety/unsound-null-safety#testing-or-running-mixed-version-programs)

nightscape, 2021-07-04:
I just merged a corresponding PR.
I can't publish new versions yet, but you can try it by using the Git version of this repo.

closed 2021-07-04 by nightscape