edufolly/flutter_bluetooth_serial #128: Find a solution for future maintenance

nightscape, 2021-05-05:
Hi @edufolly,

this project seems to be quite important to many people, but you don't seem to have time to maintain it yourself anymore.
I've seen network graphs like this before on Github, and it's a breaking point for many Open Source projects:
[image]
In almost all cases, the original author does not have the time or interest to handle the huge amount of issues, questions, pull requests, ... anymore.
It seems that there are a few people willing to help with maintaining this project (please add a 👍  here if you are one of them and actually have time and motivation to answer questions & issues, review pull requests, etc.).
It would be great if you could add contributor or admin rights to a few of them so that this project doesn't die of neglect.
You have invested quite a bit of time and energy yourself!

I would recommend e.g. @kechankrisna and @PsychoXIVI (please 👍  if you'd be willing to become a maintainer), both of whom have submitted very well-written PRs to this project.

CodeDoctorDE, 2021-05-06:
It would be cool to have a simpler api and an updated api. The latest version is updated 2019

nightscape, 2021-05-12:
@PsychoXIVI do you have sufficient rights to add additional collaborators?

edufolly, 2021-05-12:
Hi @nightscape. Please, send me an email if you want to collaborate even more.

CodeDoctorDE, 2021-06-22:
Any new updates? @edufolly @nightscape

nightscape, 2021-06-22:
Yes, I do have maintainer rights now and I'm planning to get the null-safety branch merged soon.
Just haven't found the time yet.

jahnli, 2021-06-23:
> 是的，我现在确实拥有维护者权利，并且我计划尽快合并 null-safety 分支。
> 只是还没找到时间。

Looking forward to the new maintenance, hope to migrate to air security as soon as possible

nightscape, 2021-07-04:
null-safety branch is merged, along with a few other updates and improvements.
You can try it by using the Git version of this repo.

I'm still lacking the ability to publish to pub.dev.
@edufolly @AgainPsychoX I could set up a Github action to automatically publish Github releases.
Any objections against this?

edufolly, 2021-07-05:
Version 0.3.0 published. Thanks for all the support.

closed 2021-07-05 by edufolly