edufolly/flutter_bluetooth_serial #132: Warning! Your Flutter application is created using an older version of the Android embedding. It's being deprecated in favor of Android embedding v2. Follow the steps at  https://flutter.dev/go/android-project-migration  to migrate your project.

emailsubjekt, 2021-05-27:
When I try to run your github example, I get this wonderful message 

━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━
Warning
──────────────────────────────────────────────────────────────────────────────
Your Flutter application is created using an older version of the Android
embedding. It's being deprecated in favor of Android embedding v2. Follow the
steps at

https://flutter.dev/go/android-project-migration

to migrate your project.
━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━

Launching lib\main.dart on SM G930F in debug mode...
Running Gradle task 'assembleDebug'...
√  Built build\app\outputs\flutter-apk\app-debug.apk.
Installing build\app\outputs\flutter-apk\app.apk...
Debug service listening on ws://127.0.0.1:65128/dxNEJZXiD6U=/ws
Syncing files to device SM G930F...
W/FlutterBluePlugin(25907): Local Bluetooth MAC address is hidden by system, trying other options...
D/FlutterBluePlugin(25907): Trying to obtain address using Settings Secure bank
D/FlutterBluePlugin(25907): Obtaining address using Settings Secure bank failed
D/FlutterBluePlugin(25907): Trying to obtain address using reflection against internal Android code
D/FlutterBluePlugin(25907): Probably succed: 34:8A:7B:06:C7:AF ✨ :F
V/Surface (25907): sf_framedrop debug : 0x4f4c, game : false, logging : 0
D/ViewRootImpl@6ad4639[MainActivity](25907): Relayout returned: old=[0,0][1080,1920] new=[0,0][1080,1920] result=0x7 surface={valid=true 538074120192} changed=true
I/OpenGLRenderer(25907): Initialized EGL, version 1.4
D/OpenGLRenderer(25907): Swap behavior 2
D/libGLESv1(25907): STS_GLApi : DTS, ODTC are not allowed for Package : io.github.edufolly.flutterbluetoothserialexample
D/SurfaceView(25907): BG show() Surface(name=Background for - SurfaceView - io.github.edufolly.flutterbluetoothserialexample/io.github.edufolly.flutterbluetoothserialexample.MainActivity@fd48371@0) io.flutter.view.FlutterView{fd48371 VFE...... .F....ID 0,0-1080,1920}
D/mali_winsys(25907): EGLint new_window_surface(egl_winsys_display *, void *, EGLSurface, EGLConfig, egl_winsys_surface **, egl_color_buffer_format *, EGLBoolean) returns 0x3000,  [1080x1920]-format:1
D/OpenGLRenderer(25907): eglCreateWindowSurface = 0x7d47ba2af0
V/Surface (25907): sf_framedrop debug : 0x4f4c, game : false, logging : 0
D/SurfaceView(25907): surfaceCreated 1 io.flutter.view.FlutterView{fd48371 VFE...... .F....ID 0,0-1080,1920}
D/mali_winsys(25907): EGLint new_window_surface(egl_winsys_display *, void *, EGLSurface, EGLConfig, egl_winsys_surface **, egl_color_buffer_format *, EGLBoolean) returns 0x3000,  [1080x1920]-format:1
D/SurfaceView(25907): surfaceChanged (1080,1920) 1 io.flutter.view.FlutterView{fd48371 VFE...... .F....ID 0,0-1080,1920}
D/ViewRootImpl@6ad4639[MainActivity](25907): MSG_RESIZED_REPORT: frame=Rect(0, 0 - 1080, 1920) ci=Rect(0, 72 - 0, 0) vi=Rect(0, 72 - 0, 0) or=1
D/ViewRootImpl@6ad4639[MainActivity](25907): Relayout returned: old=[0,0][1080,1920] new=[0,0][1080,1920] result=0x3 surface={valid=true 538074120192} changed=false
D/ViewRootImpl@6ad4639[MainActivity](25907): MSG_WINDOW_FOCUS_CHANGED 1
V/InputMethodManager(25907): Starting input: tba=android.view.inputmethod.EditorInfo@7ee9556 nm : io.github.edufolly.flutterbluetoothserialexample ic=null
I/InputMethodManager(25907): startInputInner - mService.startInputOrWindowGainedFocus
V/InputMethodManager(25907): Starting input: tba=android.view.inputmethod.EditorInfo@899f9d7 nm : io.github.edufolly.flutterbluetoothserialexample ic=null
D/libGLESv2(25907): STS_GLApi : DTS, ODTC are not allowed for Package : io.github.edufolly.flutterbluetoothserialexample
D/ViewRootImpl@6ad4639[MainActivity](25907): ViewPostIme pointer 0
I/zygote64(25907): Do partial code cache collection, code=30KB, data=25KB
I/zygote64(25907): After code cache collection, code=30KB, data=25KB
I/zygote64(25907): Increasing code cache capacity to 128KB
D/ViewRootImpl@6ad4639[MainActivity](25907): ViewPostIme pointer 1
D/ViewRootImpl@6ad4639[MainActivity](25907): ViewPostIme pointer 0
D/ViewRootImpl@6ad4639[MainActivity](25907): ViewPostIme pointer 1
D/ViewRootImpl@6ad4639[MainActivity](25907): ViewPostIme pointer 0
D/ViewRootImpl@6ad4639[MainActivity](25907): ViewPostIme pointer 0
D/ViewRootImpl@6ad4639[MainActivity](25907): ViewPostIme pointer 1
D/FlutterBluePlugin(25907): Starting discovery
D/BluetoothAdapter(25907): startDiscovery
I/zygote64(25907): Do partial code cache collection, code=61KB, data=48KB
I/zygote64(25907): After code cache collection, code=61KB, data=48KB
I/zygote64(25907): Increasing code cache capacity to 256KB
D/FlutterBluePlugin(25907): Discovery finished
D/BluetoothAdapter(25907): cancelDiscovery
D/BluetoothAdapter(25907): cancelDiscovery = true
D/FlutterBluePlugin(25907): Canceling discovery (stream closed)
D/BluetoothAdapter(25907): cancelDiscovery
D/BluetoothAdapter(25907): cancelDiscovery = true

### Problem summary

### Steps to reproduce

1. 
2. 
3. 
4. 

### Environment

```
```

edufolly, 2021-07-05:
Could you help? @nightscape @AgainPsychoX

nightscape, 2021-07-06:
I'm on it 👍

luckycreationsindia, 2021-08-07:
Any update on this?

It's creating issue with firebase_messaging background notifications while registering other plugins like shared_preferences (For background notification)

nightscape, 2021-08-08:
I started working on this here: https://github.com/edufolly/flutter_bluetooth_serial/tree/update_flutter_plugin_v2
Unfortunately, the plugin loading part does not seem to work on that branch and I ran out of time figuring out why.
If anybody could give this a look it would be very much appreciated!

luckycreationsindia, 2021-08-08:
I did few changes and It's working.

Debugging the code got me to null pointer on activity.

Full Changes can be found here:
https://github.com/luckycreationsindia/flutter_bluetooth_serial/tree/update_flutter_plugin_v2

This also address Future<int> issue with setState and getDeviceBondState Methods as mentioned in this issue:

#139

nightscape, 2021-08-14:
Thank you @luckycreationsindia, this was the missing part!

@edufolly would you mind creating a new release out of this?

closed 2021-08-14 by nightscape

edufolly, 2021-08-17:
Thanks, @luckycreationsindia!

And thanks @nightscape again!

Sorry about the delay, I've been published now version ```0.4.0```.

PR #144 (2021-08-14): Update flutter plugin v2
Update to the new Flutter plugin architecture v2.
Closes #132