edufolly/flutter_bluetooth_serial #133: Only Android Support?

Celpear, 2021-05-27:
Hello,
after trying your package on an IOS device, I received an error message that the IOS methods cannot be accessed. Has this problem occurred before?

### Problem summary
[image]

### Steps to reproduce

After trying the Discover function:
[image]
I get this msg:
[image]

The getState method does not work on IOS devices either.
I have also checked the IOS permissions (Info.plist). But they seem to be ok.
When searching for the methods, I couldn't find them in the native code.

### Environment

Flutter Version:
Flutter 2.0.6 • channel stable 
Framework • revision 1d9032c7e1 (4 weeks ago) • 2021-04-29 17:37:58 -0700
Engine • revision 05e680e202
Tools • Dart 2.12.3

It was installed on an IPhone SE 
IOS 14.6 installed and tested.

Can you help me further?
Thank you very much!

MojamojaK, 2021-05-29:
This library only supports Android

Celpear, 2021-06-07:
That is a pity. In pub.dev it says IOS support. :(

Moussabaddour, 2021-06-08:
unfortunately,

That plugin supports the legacy RFCOMM profile. This profile is not available to apps on iOS.

This plugin cannot work with iOS (Nor can any plugin that uses the RFCOMM profile).

edufolly, 2021-07-16:
@Celpear iOS doesn't have support. Check this issue: #2

closed 2021-07-16 by edufolly